OpenIntelWireless/IntelBluetoothFirmware #225: Bluetooth not working on AX200 card

webstar1, 2020-11-26:
Hi, I have intel bluetooth and wifi. Wifi works fine with iwltm or whatever its called but this just refuses to work, I've tried old drivers, new drivers, rearranging the loading order in opencore between the injector and firmware. It just does not work. I can see bluetooth devices on the latest release, the older releases dont even do that. However when I try to connect, it just doesnt, sometimes throws an error saying it cant connect, otherwise just says connected when it clearly isnt. The log shows the following:

2020-11-26 15:24:34.112608+0000 0x52c      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver init()
2020-11-26 15:24:34.112650+0000 0x52c      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Probe()
2020-11-26 15:24:34.112662+0000 0x52c      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0029
2020-11-26 15:24:34.112772+0000 0x52c      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Start()
2020-11-26 15:24:35.615631+0000 0x52c      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: set configuration to 1
2020-11-26 15:24:35.620545+0000 0x52c      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found interface!!!
2020-11-26 15:24:35.620590+0000 0x52c      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found Interrupt endpoint!
2020-11-26 15:24:35.622759+0000 0x52c      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found Bulk out endpoint!
2020-11-26 15:24:35.624806+0000 0x52c      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found Bulk in endpoint!
2020-11-26 15:24:35.627482+0000 0x52c      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: usb init succeed
2020-11-26 15:24:35.627485+0000 0x52c      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2020-11-26 15:24:35.630228+0000 0x52c      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-11-26 15:24:35.631234+0000 0x166      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Firmware revision 0.0 build 250 week 17 2020
2020-11-26 15:24:35.631238+0000 0x166      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: suspect device firmware: ibt-20-0-0.sfi
2020-11-26 15:24:35.631241+0000 0x166      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: firmware had been download.
2020-11-26 15:24:35.631245+0000 0x52c      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-11-26 15:24:35.631247+0000 0x52c      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_EVENT_MASK
2020-11-26 15:24:35.635233+0000 0x52c      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-11-26 15:24:35.636222+0000 0x52c      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-11-26 15:24:35.636228+0000 0x52c      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: End download
2020-11-26 15:24:35.636230+0000 0x52c      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Clean up...

williambj1, 2020-11-26:
Connection issues are not related to this kext.

closed 2020-11-26 by williambj1

rickgerhart, 2021-06-07:
I have two almost identical Hackintoshes each with different AX200 cards running BigSur. Both demonstrate  this same load error. I've tried all the tricks  from software versions to switching internal USB ports but the glitch remained. Until I installed a external USB Bluetooth adapter and unplugged the internal wifi/bluetooth from its bluetooth connection,  not the solution I wanted but ...