OpenIntelWireless/IntelBluetoothFirmware #226: Cannot connect G613 Bluetooth keyboard

MarkoBogSrb, 2020-11-26:
**Description**
When I put the keyboard in the pairing mode, and I go into sys preferences -> Keyboard -> set up Bluetooth keyboard, and I click on the connect, it loads for about a minute until it just refreshes and does not connect. I also tried pairing via Bluetooth options and it did not work. (However, I can pair my random headphones no problem but there is some lag.)

**Environment**

- Kext Version: Latest (1.2.0)
- WiFi Card Model: AC7260
- Product ID: 0x8087, 0x07dc
- macOS Version: Big Sur 11.0.1

**Info in System Report - Bluetooth**

```
Apple Bluetooth Software Version:	8.0.1f5
  Hardware, Features, and Settings:
  Name:	Marko’s MacBook Pro
  Address:	7C-5C-F8-A8-B3-B1
  Bluetooth Low Energy Supported:	Yes
  Handoff Supported:	Yes
  Instant Hot Spot Supported:	Yes
  Manufacturer:	Intel
  Transport:	USB
  Firmware Version:	6.3584 (6.3584)
  Bluetooth Power:	On
  Discoverable:	Off
  Connectable:	Yes
  Auto Seek Pointing:	On
  Remote wake:	On
  Vendor ID:	0x8087
  Product ID:	0x07DC
  Bluetooth Core Spec:	4.0 (0x6)
  HCI Revision:	0xE00
  LMP Version:	4.0 (0x6)
  LMP Subversion:	0xE00
  Device Type (Major):	Computer
  Device Type (Complete):	Mac Portable
  Composite Class Of Device:	0x38010C
  Device Class (Major):	0x01
  Device Class (Minor):	0x03
  Service Class:	0x1C0
  Auto Seek Keyboard:	On
  Devices (Paired, Configured, etc.):
  MPOW H19 IPO:
  Address:	FC-58-FA-05-1E-E0
  Major Type:	Audio
  Minor Type:	Headphones
  Services:	Hands-Free unit, Headset, Headset, CSR GAIA‚Ñ¢
  Paired:	Yes
  Configured:	Yes
  Connected:	No
  Manufacturer:	Cambridge Silicon Radio (0x9, 0x3261)
  Bluetooth Core Spec:	5.0
  Firmware Version:	0xFFFF
  Vendor ID:	0x000A
  Product ID:	0xFFFF
  Class of Device:	0x04 0x06 0x240418
  Host Connectable:	No
  EDR Supported:	Yes
  eSCO Supported:	Yes
  SSP Supported:	Yes
  Services:
  Bluetooth File Transfer:
  Folder other devices can browse:	~/Public
  When receiving items:	Accept all without warning
  State:	Disabled
  Bluetooth File Exchange:
  Folder for accepted items:	~/Downloads
  When other items are accepted:	Save to location
  When receiving items:	Accept all without warning
  State:	Disabled
  Bluetooth Internet Sharing:
  State:	Disabled
  Incoming Serial Ports:
  Bluetooth-Incoming-Port:
  RFCOMM Channel:	3
  Requires Authentication:	No
  Outgoing Serial Ports:
  MPOWH19IPO-CSRGAIA:
  Address:	FC-58-FA-05-1E-E0
  RFCOMM Channel:	1
  Requires Authentication:	No
```

**Screenshots**

[image]
[image]

williambj1, 2020-11-26:
Connection issues are not related to firmware uploading. Read our docs for more info.

williambj1, 2020-11-26:
Duplicate of #51

closed 2020-11-26 by williambj1