OpenIntelWireless/IntelBluetoothFirmware #227: If you have pair/connection issue, make sure you connect your antenna "properly"

shasen1235, 2020-11-27:
**Description**
Not a bug report what so ever, but I think this experience can help people with pairing or connecting issue.

**Environment**

- Kext Version: 1.1.2
- WiFi Card Model: AX200 rev.3(0C)
- Product ID: 
- macOS Version: 10.14.6
- Tested machine: Dell XPS 9570

**Screenshots**
Long story short, if you came from a killer card and trying to install a new intel card. Check your main&aux antenna first.
[image]

See the problem here? The white indicator on the killer card is for "Main" and the black is for "Aux" while on the Intel side is totally opposite, which is kind of a...bummer...

Here's another photo I took before I did the installation and it clearly shows that the killer card has the correct order.
[image]

Funny thing is that, if you connect the card in wrong order, wifi will still work, at least it provided 300+Mbps on speed test.
However, on the bluetooth side is a totally different story, I can barely find any device near my laptop. After since I corrected the order, it pairs my wireless headphone instantly and play through a 30 min youtube video without any stutter or disconnect what so ever.

So the bottom line, do not believe the color of the indicators, find the proper antenna for the port it truly responds to.

Hope this will help.
J

williambj1, 2020-11-29:
Thanks for your guide

korawit1188, 2022-04-23:
Ok 't