OpenIntelWireless/IntelBluetoothFirmware #234: 关于更换任意机型蓝牙模块不被识别到问题

albert97chn, 2020-12-11:
首先感谢此Kext，让我的小米游戏本在BigSur 11.0.1 （OC0.6.3） 中的驱动蓝牙模块（ INTEL AC9560 **0x8087, 0x0aaa** ），可以正常使用蓝牙设备，也可通过蓝牙联网，体验几乎完美。
但是这个只局限于机型被设置为_MacBook Pro (13-inch, 2018, Four Thunderbolt 3 Ports)_时才行，设置为其他任意机型都无法识别该设备。测试过2015-2020**所有机型**，包括各尺寸Pro Air Mac Pro Macmini等等（_测了近9个小时快吐了🤮_）。
我不确定是否是该Kext的问题，甚至认为有问题的是**IOUSB Host Device**。

[image]

上图是在MacBook Pro (13-inch, 2018, Four Thunderbolt 3 Ports)机型下Hackintool-USB中的截图，红框内即为我的蓝牙设备**INTEL AC9560 0x8087, 0x0aaa ** ，被单独识别在上面的意思虽然不知道，但是在下方USB设备列表内是可以被识别到的（名称HS14 接口0*0E）。在系统报告中 硬件-USB 也是可以被识别到，如图：
[image]
蓝牙中更是没有问题。

但是当设置机型为任意其他时，蓝牙模块不会被识别，在Hackintool中，上栏依然正常显示，显然是被系统识别到了。但是下栏USB接口却没有该设备。在系统报告 硬件-USB 也不会出现“蓝牙HCI”，蓝牙显示没有设备。打开系统偏好设置，甚至没有蓝牙控制面板。

经过学习和琢磨，我进行了对USB驱动的定制，和取消定制。也重新仔细阅读了OpenCore 0.6.3的引导配置说明，虽然解决了一些其他驱动的显示和BUG。但是蓝牙模块在其他机型不可驱动的问题始终存在。

如果完全是USB总线的问题，为何同为一体的_Intel AC9560 无线网卡_可以被驱动，其他USB，无论内建还是外置接口也都不存在问题。之后再交流群中看到有人提到，蓝牙限制机型。就想求问作者，蓝牙模块是否限定机型，又如何解决换机型不被驱动的问题。

希望作者能够指点解惑。感谢🙏

williambj1, 2020-12-11:
USBPorts 定制 USB 仅针对 plist 中写入的机型有效，请在更改 smbios 之后同步更改 plist 中的机型，谢谢

closed 2020-12-11 by williambj1

albert97chn, 2020-12-18:
感谢指点 
在Kext-USBPorts-Contents-Info.plist手动添加了无法识别的USB端口，现在蓝牙端口已被驱动