OpenIntelWireless/IntelBluetoothFirmware #239: 蓝牙导致睡眠无法唤醒

ghsssss, 2020-12-27:
**Description**

**Environment**

- Kext Version: 
- WiFi Card Model: 
- Product ID: 
- macOS Version: 

**Info in System Report - Bluetooth**

```

```

**Logs**

```log

```

**Kernel Panic Logs**

```

```

**Screenshots**

ghsssss, 2020-12-27:
panic(cpu 4 caller 0xffffff8010e8915a): Wake transition timed out after 180 seconds while calling power state change callbacks. Suspected bundle: com.apple.iokit.IOBluetoothHostControllerTransport. Thread 0x11216.
Failure code:: 0x00010004 00000027

Backtracing panicked thread because kernel_stack is NULL for specified thread
Backtrace (CPU 4), Frame : Return Address
0xffffffa0a7563880 : 0xffffff80106b9aed 
0xffffffa0a75638d0 : 0xffffff80107fc6e3 
0xffffffa0a7563910 : 0xffffff80107ecd1a 
0xffffffa0a7563960 : 0xffffff801065ea2f 
0xffffffa0a7563980 : 0xffffff80106b938d 
0xffffffa0a7563aa0 : 0xffffff80106b9678 
0xffffffa0a7563b10 : 0xffffff8010ebe4c8 
0xffffffa0a7563b80 : 0xffffff8010e8915a 
0xffffffa0a7563e60 : 0xffffff8010dfbe4a 
0xffffffa0a7563e80 : 0xffffff80106fe6b5 
0xffffffa0a7563ef0 : 0xffffff80106ff5c4 
0xffffffa0a7563fa0 : 0xffffff801065e13e 

Process name corresponding to current thread: kernel_task
Boot args: darkwake=1 

Mac OS version:
20C69

Kernel version:
Darwin Kernel Version 20.2.0: Wed Dec  2 20:39:59 PST 2020; root:xnu-7195.60.75~1/RELEASE_X86_64
Kernel UUID: 82E2050C-5936-3D24-AD3B-EC4EC5C09E11
KernelCache slide: 0x0000000010400000
KernelCache base:  0xffffff8010600000
Kernel slide:      0x0000000010410000
Kernel text base:  0xffffff8010610000
__HIB  text base: 0xffffff8010500000
System model name: MacBookPro13,3 (Mac-A5C67F76ED83108C)
System shutdown begun: NO
Panic diags file available: YES (0x0)
Hibernation exit count: 0

System uptime in nanoseconds: 8149875205923
Last Sleep:           absolute           base_tsc          base_nano
  Uptime  : 0x000007698a6aaa9f
  Sleep   : 0x000006fcb9efae10 0x00000000b776dbf2 0x000006f10962cc6f
  Wake    : 0x000006fce4a83513 0x00000000bad963bc 0x000006fcba4baec7
last started kext at 5876921129108: >!UAudio	401.4 (addr 0xffffff7fb0b0f000, size 307200)
loaded kexts:
com.zxystd.!I!BFirmware	1.1.2
me.kishorprins.VoodooInput	1.1.0
com.alexandred.VoodooI2CHID	1
as.acidanthera.voodoo.driver.PS2Keyboard	2.1.9
com.sinet3k.Sinetek-rtsx	7
com.zxystd.AirportItlwm	1.2.0
org.rehabman.driver.FakePCIID	1.3.15
com.alexandred.VoodooI2C	2.6.1
org.coolstar.VoodooGPIO	1.1
as.acidanthera.voodoo.driver.PS2!C	2.1.9
as.vit9696.SMCProcessor	1.1.9
as.vit9696.WhateverGreen	1.4.6
com.XPS.VerbStub	1.0.3
as.lvs1974.SMCDellSensors	1.1.9
ru.usrsse2.SMCBatteryManager	1.1.9
as.vit9696.VirtualSMC	1.1.9
org.acidanthera.NVMeFix	1.0.4
as.acidanthera.BrightnessKeys	1.0.1
com.alexandred.VoodooI2CServices	1
as.vit9696.!AALC	1.5.5
org.vanilla.driver.CPUFriend	1.2.2
as.vit9696.Lilu	1.5.0
>AudioAUUC	1.70
>!AHDAHardwareConfigDriver	283.15
>!AHDA	283.15
>!APlatformEnabler	2.7.0d0
>AGPM	119
>X86PlatformShim	1.0.0
@fileutil	20.036.15
@filesystems.autofs	3.0
@filesystems.ntfs	3.14.3
>!AGraphicsDevicePolicy	6.2.2
@AGDCPluginDisplayMetrics	6.2.2
>!AUpstreamUserClient	3.6.8
>pmtelemetry	1
>LuaHardwareAccess	1.0.16
|IOUserEthernet	1.0.1
>usb.!UUserHCI	1
|IO!BSerialManager	8.0.2f9
>!A!IPCHPMC	2.0.1
@Dont_Steal_Mac_OS_X	7.0.0
>!AHV	1
>!ADiskImages2	1
>AGDCBacklightControl	6.2.2
>!A!ISKLGraphics	16.0.1
>!ABacklight	180.3
>!AFIVRDriver	4.1.0
>ACPI_SMC_PlatformPlugin	1.0.0
>!A!ISlowAdaptiveClocking	4.0.0
>!A!ISKLGraphicsFramebuffer	16.0.1
>!AMCCSControl	1.14
>!AFileSystemDriver	3.0.1
@filesystems.tmpfs	1
@filesystems.hfs.kext	556.60.1
@BootCache	40
@!AFSCompression.!AFSCompressionTypeZlib	1.0.0
@!AFSCompression.!AFSCompressionTypeDataless	1.0.0d1
>!ASmartBatteryManager	161.0.0
>!ATopCaseHIDEventDriver	4020.6
@filesystems.apfs	1677.60.23
>!AAHCIPort	346
@private.KextAudit	1.0
>!ARTC	2.0
>!AACPIEC	6.1
>!AACPIButtons	6.1
>!AHPET	1.8
>!ASMBIOS	2.1
>!AAPIC	1.7
@!ASystemPolicy	2.0.0
@nke.applicationfirewall	310
|IOKitRegistryCompatibility	1
|EndpointSecurity	1
>!UAudio	401.4
>DspFuncLib	283.15
@kext.OSvKernDSPLib	529
@kext.triggers	1.0
>!AGraphicsControl	6.2.2
|IOSerial!F	11
|IOAVB!F	900.12
>usb.IOUSBHostHIDDevice	1.2
>!ABacklightExpert	1.1.0
|IONDRVSupport	585
>!AHDA!C	283.15
|IOHDA!F	283.15
|IOAudio!F	300.6.1
@vecLib.kext	1.2.0
@!AGPUWrangler	6.2.2
>IOPlatformPluginLegacy	1.0.0
>X86PlatformPlugin	1.0.0
>IOPlatformPlugin!F	6.0.0d8
|IOSlowAdaptiveClocking!F	1.0.0
|IOAccelerator!F2	439.52
@!AGraphicsDeviceControl	6.2.2
|IOGraphics!F	585
>!ASMBus!C	1.0.18d1
>!ASMBusPCI	1.0.14d1
@plugin.IOgPTPPlugin	900.11
|IOEthernetAVB!C	1.1.0
|Broadcom!BHost!CUSBTransport	8.0.2f9
|IO!BHost!CUSBTransport	8.0.2f9
|IO!BHost!CTransport	8.0.2f9
>usb.cdc	5.0.0
>usb.networking	5.0.0
>usb.!UHostCompositeDevice	1.2
>!UMergeNub	900.4.2
>!ABSDKextStarter	3
|IOSurface	289.3
@filesystems.hfs.encodings.kext	1
>usb.!UHostPacketFilter	1.0
|IOUSB!F	900.4.2
>!AActuatorDriver	4400.28
>!AMultitouchDriver	4400.28
>!AInputDeviceSupport	4400.35
>!AHS!BDriver	4020.6
>IO!BHIDDriver	8.0.2f9
>!AXsanScheme	3
|IONVMe!F	2.1.0
|IO80211!F	1200.12.2b1
|IOSkywalk!F	1
>mDNSOffloadUserClient	1.0.1b8
>corecapture	1.0.4
|IOAHCI!F	294
>!A!ILpssI2C	3.0.60
>!A!ILpssGspi	3.0.60
>usb.!UXHCIPCI	1.2
>usb.!UXHCI	1.2
>!AEFINVRAM	2.1
>!AEFIRuntime	2.1
$!AImage4	3.0.0
|IOSMBus!F	1.1
|IOHID!F	2.0.0
|IOTimeSync!F	900.11
|IONetworking!F	3.4
>DiskImages	493.0.0
|IO!B!F	8.0.2f9
|IOReport!F	47
|IO!BPacketLogger	8.0.2f9
$quarantine	4
$sandbox	300.0
@kext.!AMatch	1.0.0d1
|CoreAnalytics!F	1
>!ASSE	1.0
>!AKeyStore	2
>!UTDM	511.60.2
|IOUSBMass!SDriver	184.40.6
|IOSCSIBlockCommandsDevice	436.40.6
|IO!S!F	2.1
|IOSCSIArchitectureModel!F	436.40.6
>!AMobileFileIntegrity	1.0.5
@kext.CoreTrust	1
>!AFDEKeyStore	28.30
>!AEffaceable!S	1.0
>!ACredentialManager	1.0
>KernelRelayHost	1
|IOUSBHost!F	1.2
>!UHostMergeProperties	1.2
>usb.!UCommon	1.0
>!ABusPower!C	1.0
>!ASEPManager	1.0.1
>IOSlaveProcessor	1
>!AACPIPlatform	6.1
>!ASMC	3.1.9
|IOPCI!F	2.9
|IOACPI!F	1.4
>watchdog	1
@kec.pthread	1
@kec.corecrypto	11.1
@kec.Libm	1

ghsssss, 2020-12-27:
请问是这个驱动的问题吗？

williambj1, 2020-12-27:
https://github.com/acidanthera/OpenCorePkg/blob/3869e19f1d67def1b392c57cc810cee9c44a5073/Docs/Sample.plist#L754-L755

==> `true`

closed 2020-12-27 by williambj1

ghsssss, 2020-12-27:
0.0这个选项一直是true

ghsssss, 2020-12-27:
@williambj1