OpenIntelWireless/IntelBluetoothFirmware #242: Bluetooth is not working at all and apple logo loading screen takes about 5 minutes

borasumer, 2021-01-05:
**Description**
Looks like it does not load the bluetooth kext. It takes about 5 mins to load up and start the pc, normally it was like 10 seconds since I have an ssd. But for the next couple of days, after I shut the pc down and start again. It takes 5 mins to load, and after that bluetooth is not working at all. Couple of times restarting helped, but now it does not work either. And yesterday I could not solve the issue and wanted to restore with migration assistant, and realized that I had only older backups, so in the middle of migration selection menu I quit the migration assistant, and all of a sudden bluetooth started working. Because I think migration assistant closes all the apps and services, so technically it restarted the services thus it loaded the kexts again. Not sure what happened to be honest but it seems like this is the only way to get it work, but it is painful to wait 5 mins every time after I start the pc. 

- Kext Version: 
- 1.2
- WiFi Card Model: 
- ac3160
- macOS Version: 
Big Sur 11.1

system logs shows nothing

jgyjiayou, 2021-01-06:
无药可救

hiby-arch, 2021-01-10:
I have the same problem.  After maybe half an hour the Bluetooth will get into work(which can be seen in the status bar). Sometimes it works if I rebuild the kext cache. I thought I might be the only one who have this issue until I find your description...

williambj1, 2021-01-16:
Sounds weird and seems to be a system or user configuration issue, does your Bluetooth work properly if you do a warm reboot from Windows?

VergeDX, 2021-01-16:
Well... Seems I meet this issues too. In my Big Sur 11.1, I can see IntelFramework's log, it print the same content again and again, my solution is not load IntelBluetoothInjector.kext and wait to fix.
If any log needed, please let me know : )

bolu-atx, 2021-01-22:
> Sounds weird and seems to be a system or user configuration issue, does your Bluetooth work properly if you do a warm reboot from Windows?

After rebooting into windows the bluetooth issue resolved itself, do you know how to do a permanent fix without rebooting into windows every time?

zxystd, 2021-06-29:
Try the latest alpha version. Close since this issue is inactive. Thanks.

closed 2021-06-29 by zxystd