OpenIntelWireless/IntelBluetoothFirmware #245: Connection very unstable (Yes all the connections in my laptop are fine)

linusrg1, 2021-01-15:
**Description**

When I connect a pair of bluetooth headphones I can only be a few inches away before it starts breaking up.

**Environment**

- Kext Version: 1.1.2
- WiFi Card Model: 7265AC
- Product ID: 8086 095b
- macOS Version: Big Sur 11

**Info in System Report - Bluetooth**

```
  Apple Bluetooth Software Version:	8.0.3d8
  Hardware, Features, and Settings:
  Address:	18-5E-0F-AF-05-7C
  Bluetooth Low Energy Supported:	Yes
  Handoff Supported:	Yes
  Instant Hot Spot Supported:	Yes
  Manufacturer:	Intel
  Transport:	USB
  Firmware Version:	8.4096 (8.4096)
  Bluetooth Power:	Off
  Auto Seek Pointing:	On
  Remote wake:	On
  Vendor ID:	0x8087
  Product ID:	0x0A2A
  Bluetooth Core Spec:	4.2 (0x8)
  HCI Revision:	0x1000
  LMP Version:	4.2 (0x8)
  LMP Subversion:	0x1000
  Auto Seek Keyboard:	On
  Devices (Paired, Configured, etc.):
  Mpow M30:
  Address:	00-00-AB-CF-8D-3F
  Major Type:	Audio
  Minor Type:	Headset
  Services:	SerialPort, Hands-Free unit
  Paired:	No
  Configured:	Yes
  Connected:	No
  Manufacturer:	Unknown (0x9, 0x3232)
  Bluetooth Core Spec:	5.0
  Class of Device:	0x04 0x01 0x240404
  EDR Supported:	Yes
  eSCO Supported:	Yes
  SSP Supported:	Yes
  Services:
  Bluetooth File Transfer:
  Folder other devices can browse:	~/Public
  When receiving items:	Accept all without warning
  State:	Enabled
  Bluetooth File Exchange:
  Folder for accepted items:	~/Downloads
  When other items are accepted:	Save to location
  When receiving items:	Accept all without warning
  State:	Enabled
  Bluetooth Internet Sharing:
  State:	Disabled
  Incoming Serial Ports:
  Bluetooth-Incoming-Port:
  RFCOMM Channel:	3
  Requires Authentication:	No
  Outgoing Serial Ports:
  MpowM30-SerialPort:
  Address:	00-00-AB-CF-8D-3F
  RFCOMM Channel:	3
  Requires Authentication:	No

```

**Logs**

```log
nothing appeared in log
```

**Kernel Panic Logs**

```

```

**Screenshots**

williambj1, 2021-01-16:
7000 series cards are known to suffer from serious interference, are you using 2.4 GHz at the same time?

Closing since this issue is not related to this kext, please read our docs for more info.

closed 2021-01-16 by williambj1

linusrg1, 2021-01-16:
My bluetooth worked fine on linux however, and I am on 5ghz wifi.