OpenIntelWireless/IntelBluetoothFirmware #247: 编译出错 `Error: Process completed with exit code 65.`

70599, 2021-01-18:
[参考这里](https://openintelwireless.github.io/IntelBluetoothFirmware/Advanced.html#build-from-source)编译时，出现以下错误：
```
xcodebuild: error: The flag -scheme or -xctestrun is required when specifying -derivedDataPath.
```

williambj1, 2021-01-18:
什么 xcode 版本

70599, 2021-01-18:
我是fork一份，然后用workflow尝试的。
```yaml
name: Build

on:
  push:
    branches: [ master ]
  pull_request:
    branches: [ master ]

jobs:
    build:

      runs-on: macos-latest

      steps:
        - name: Checkout
          uses: actions/checkout@v2

        - name: Build
          env:
            PROJECT_DIR: .
          run: |
            git clone --depth=1 https://github.com/acidanthera/MacKernelSDK.git
            xcodebuild -alltargets -derivedDataPath build
        - name: Upload assets
          uses: actions/upload-artifact@v2
          with:
            path:
              ./build/Build/Products/Release
```

williambj1, 2021-01-18:
可以参考仓库内的 workflow

70599, 2021-01-18:
谢谢，但是仓库里的`main.yml`没有`upload-artifact`不知道该怎样取得kext。
请问仓库里的`main.yml`编译后，kext输出路径是哪里？

70599, 2021-01-18:
我先用`find`尝试下

70599, 2021-01-18:
得到kext了，感谢帮助。

closed 2021-01-18 by 70599