OpenIntelWireless/IntelBluetoothFirmware #248: Bluetooth kext not loading at all on Catalina (wifi kexts are working)

bolu-atx, 2021-01-18:
**Description**
I downloaded the latest built kexts v1.1.2 and couldn't get Bluetooth to work on my Hacknitosh
Tried the suggestion to check boot logs, but nothing referring to IntelBluetoothFirmware even shows up

```
~ » log show --last boot | grep Intel                                      blu@Bos-iMac-Pro
2021-01-18 11:44:36.479243-0800 0x62f      Default     0x0                  0      0    kernel: (AppleIntelMCEReporter) AppleIntelMCE:Controller: This platform does not support CMCI
2021-01-18 11:44:42.534825-0800 0x2b5      Default     0x0                  0      0    kernel: (IOPlatformPluginFamily) ACPI_SMC_PlatformPlugin::start - waitForService(resourceMatching(AppleIntelCPUPowerManagement) timed out
2021-01-18 11:44:42.534827-0800 0x2b5      Default     0x0                  0      0    kernel: (IOPlatformPluginFamily) ACPI_SMC_PlatformPlugin::start - waitForService(resourceMatching(AppleIntelCPUPowerManagement) timed out
Can't get value for key 'AppleIntelFramebuffer'
2021-01-18 11:45:07.878230-0800 0xda0      Default     0x0                  344    2    lsd: (LaunchServices) [com.apple.launchservices:default] - 45683955: Checking whether application is managed at file:///Applications/HeliPort.app/Contents/Library/LoginItems/HeliPort%20Launcher.app/com.OpenIntelWireless.HeliPort-Launcher
        "com.OpenIntelWireless.HeliPort-Launcher"
2021-01-18 11:45:07.972587-0800 0xd68      Default     0x0                  339    0    otherbsd: Could not submit LoginItem job com.OpenIntelWireless.HeliPort-Launcher: 119: Service is disabled
2021-01-18 11:45:09.689317-0800 0xfd3      Default     0x0                  358    0    Dock: (DeviceManagement) [com.apple.dmd:Policy] Requested application <private> has policy 0, associated category:DH1009 associated sites:(null) equivalent bundle identifiers:com.OpenIntelWireless.HeliPort
2021-01-18 11:45:09.741055-0800 0xef4      Default     0x0                  360    0    Finder: (DeviceManagement) [com.apple.dmd:Policy] Requested application <private> has policy 0, associated category:DH1009 associated sites:(null) equivalent bundle identifiers:com.OpenIntelWireless.HeliPort-Launcher
(base) ---
```

Looks like the wiresless kext is working fine

**Environment**

- Kext Version:  1.1.2
- WiFi Card Model:  Intel Wifi+Bluetooth 4.2 ( AsRock b450 mini-itx with Ryzen 3700X CPU)
- Product ID: 0x8087, 0x0aa7
- macOS Version:  10.15

**Kexts**
[image]

**configs.plist** Relevant Section

```
                        <dict>
                                <key>Comment</key>
                                <string>intel wifi support</string>
                                <key>MaxKernel</key>
                                <string></string>
                                <key>PlistPath</key>
                                <string>Contents/Info.plist</string>
                                <key>Enabled</key>
                                <true/>
                                <key>MinKernel</key>
                                <string></string>
                                <key>ExecutablePath</key>
                                <string>Contents/MacOS/AirportItlwm</string>
                                <key>Arch</key>
                                <string>Any</string>
                                <key>BundlePath</key>
                                <string>AirportItlwm.kext</string>
                        </dict>
                        <dict>
                                <key>Comment</key>
                                <string>intel bluetooth firmware</string>
                                <key>MaxKernel</key>
                                <string></string>
                                <key>PlistPath</key>
                                <string>Contents/Info.plist</string>
                                <key>Enabled</key>
                                <true/>
                                <key>MinKernel</key>
                                <string></string>
                                <key>ExecutablePath</key>
                                <string>Contents/MacOS/IntelBluetoothFirmware</string>
                                <key>Arch</key>
                                <string>Any</string>
                                <key>BundlePath</key>
                                <string>IntelBluetoothFirmware.kext</string>
                        </dict>
                        <dict>
                                <key>Comment</key>
                                <string>intel bluetooth firmware injector</string>
                                <key>MaxKernel</key>
                                <string></string>
                                <key>PlistPath</key>
                                <string>Contents/Info.plist</string>
                                <key>Enabled</key>
                                <true/>
                                <key>MinKernel</key>
                                <string></string>
                                <key>ExecutablePath</key>
                                <string>Contents/MacOS/IntelBluetoothInjector</string>
                                <key>Arch</key>
                                <string>Any</string>
                                <key>BundlePath</key>
                                <string>IntelBluetoothInjector.kext</string>
                        </dict>
```

Has anyone run into a similar issue before?

sourceblaak, 2021-01-18:
Yes, I'm having the same issue. Intel AX200, Catalina 10.15.7

williambj1, 2021-01-19:
@sourceblaak Your issue is different and that has nothing to do with this kext, please read our docs for more info.

@bolu-atx `IntelBluetoothFirmwareInjector.kext` is a plist-only kext and does not contain any binary files, the configuration is incorrect.

Could you post the output from kextstat?

bolu-atx, 2021-01-19:
@williambj1  Kextstat shows that the kernel extension is not loaded at all
```
~ » kextstat | grep Intel                                                  blu@Bos-iMac-Pro
   44    0 0xffffff7f83cd1000 0x33000    0x33000    com.apple.driver.AppleIntelCPUPowerManagement (222.0.0) B353FC17-B553-3561-8849-94AD2E4F828C <8 7 6 5 3 1>
  110    0 0xffffff7f85e20000 0x3000     0x3000     com.apple.driver.AppleIntelSlowAdaptiveClocking (4.0.0) 37A296C5-D4AF-3777-A3F7-3EDD1402626A <109 5 3>
  111    0 0xffffff7f85e23000 0x13000    0x13000    com.apple.driver.AppleIntelMCEReporter (115) 62FC776D-75D3-365A-A6F2-A5CB65314F5E <13 12 8 7 6 5 3 1>
```

The AirportItlwm is loaded, however
```
~ » kextstat | grep zxystd                                                 blu@Bos-iMac-Pro
   76    0 0xffffff7f85001000 0xd3a000   0xd3a000   com.zxystd.AirportItlwm (1.2.0) 9EE740EA-AB63-3492-A914-0AD5017E8DF5 <75 32 13 6 5 3 1>
```

bolu-atx, 2021-01-19:
I managed to get this working - turned out this is because of my USB mapping SSDT being configured incorrectly. After updating the SSDT to the following (Specifically for Asorck B450 mini-ITX board. your mileage might vary), the kext loaded no problem.

kextstat shows the proper kernel extension!
```
   90    0 0xffffff7f84135000 0xecc000   0xecc000   com.zxystd.IntelBluetoothFirmware (1.1.2) 7D93D8C7-7FA8-3224-AE35-8DFFC844BB63 <21 6 5 3>
```

Here's my DSL file (check this out if you don't know how to use it - https://dortania.github.io/Getting-Started-With-ACPI/Manual/compile.html#macos)
```
DefinitionBlock ("SSDT-ASROCK B450 ITX USB.aml", "SSDT", 1, "APPLE", "tinySSDT", 0x00000006)
{
    External (\_SB_.PCI0.GPP2, DeviceObj)
    External (\_SB_.PCI0.GPP2.PTXH, DeviceObj)

    Scope (\_SB_.PCI0.GPP2.PTXH)
    {
        Name (_STA, Zero)
    }

    Scope (\_SB_.PCI0.GPP2)
    {
        Device (XHCI)
        {
            Name (_ADR, Zero)
            
            Device (RHUB)
            {
                Name (_ADR, Zero)
                
                // USB3.1 back panel
                Device (PRT1)
                {
                    Name (_ADR, 0x5)
                            
                    Name (_UPC, Package (0x04)
                    {
                        0xFF, 
                        0x03, 
                        Zero, 
                        Zero
                    })
                }
                Device (PRT2)
                {
                    Name (_ADR, 0x6)
                            
                    Name (_UPC, Package (0x04)
                    {
                        0xFF, 
                        0x03, 
                        Zero, 
                        Zero
                    })
                }
                
                // USB3.0 front panel
                Device (PRT3)
                {
                    Name (_ADR, 0x7)
                            
                    Name (_UPC, Package (0x04)
                    {
                        0xFF, 
                        0x03, 
                        Zero, 
                        Zero
                    })
                }
                Device (PRT4)
                {
                    Name (_ADR, 0x8)
                            
                    Name (_UPC, Package (0x04)
                    {
                        0xFF, 
                        0x03, 
                        Zero, 
                        Zero
                    })
                }
                
                // USB2.0
                Device (PRT5)
                {
                    Name (_ADR, 0x9)
                            
                    Name (_UPC, Package (0x04)
                    {
                        0xFF, 
                        Zero, 
                        Zero, 
                        Zero
                    })
                }
                Device (PRT6)
                {
                    Name (_ADR, 0xa)
                            
                    Name (_UPC, Package (0x04)
                    {
                        0xFF, 
                        Zero, 
                        Zero, 
                        Zero
                    })
                }
                Device (PRT7)
                {
                    Name (_ADR, 0xb)
                            
                    Name (_UPC, Package (0x04)
                    {
                        0xFF, 
                        Zero, 
                        Zero, 
                        Zero
                    })
                }
                Device (PRT8)
                {
                    Name (_ADR, 0xc)
                            
                    Name (_UPC, Package (0x04)
                    {
                        0xFF, 
                        Zero, 
                        Zero, 
                        Zero
                    })
                }
                Device (PRT9)
                {
                    Name (_ADR, 0xd)
                            
                    Name (_UPC, Package (0x04)
                    {
                        0xFF, 
                        Zero, 
                        Zero, 
                        Zero
                    })
                }
                Device (PRTA)
                {
                    Name (_ADR, 0xe)
                            
                    Name (_UPC, Package (0x04)
                    {
                        0xFF, 
                        Zero, 
                        Zero, 
                        Zero
                    })
                }
            }    

        }
    }
}

```

bolu-atx, 2021-01-20:
Closing this issue. Thanks!

closed 2021-01-20 by bolu-atx