OpenIntelWireless/IntelBluetoothFirmware #249: Bluetooth kext not loading (Big Sur)

TimLampen, 2021-01-20:
Hello,

I have been trying over the past couple days to get my kext to be loaded. However, there seems to be no output to the console during start up even though all my other kexts (including Airport) are fully loaded.

This is again confirmed by using `log show --last boot | grep IntelFirmware` where there was no output.

Here is a picture of it inside my config.plist:
[image]

I have also noticed that when I use USBMap, my Bluetooth USB controller is not present (although it does work on Windows).

Thanks for the support,
Tim

bolu-atx, 2021-01-20:
Looks like a similar issue to the one I experienced: https://github.com/OpenIntelWireless/IntelBluetoothFirmware/issues/248

I recommend you search for a SSDT that works for your motherboard chipset / or go through the USB mapping process carefully following the OpenCore guides.

TimLampen, 2021-01-20:
> Looks like a similar issue to the one I experienced: #248
> 
> I recommend you search for a SSDT that works for your motherboard chipset / or go through the USB mapping process carefully following the OpenCore guides.

Hi bolu, thanks for the reply. My problem is that I am unable to map the USB in macOS as it doesn’t not appear in any of the tools I have tried to use.

additionaly, because of my laptop (comet lake) my PCI / EC name fits the default file perfectly, there is nothing to change.

TimLampen, 2021-01-20:
Solved by including USB RHUB as ACPI File.

closed 2021-01-20 by TimLampen