OpenIntelWireless/IntelBluetoothFirmware #252: 期待蓝牙v1.3

ahaosuan, 2021-01-25:
目前支持的蓝牙鼠标还是很少 而且如果在连接蓝牙失败后
wifi 会短时间断连

目前itlwm已经用上了v1.3。wifi 非常舒服

期待蓝牙v1.3

期待作者，也很感谢作者对黑苹果的研究

zxystd, 2021-01-25:
感谢你的支持，不过目前我全身心把空余时间都投入到Wi-Fi开发上了，对于蓝牙的研究只能暂停，目前来看蓝牙兼容性问题并不是这个驱动的问题，而是就像WEG对显卡的修补一个道理，并不能直接在macOS上使用第三方的网卡，因为苹果本身就没有做这方面的兼容。

ahaosuan, 2021-01-25:
> 感谢你的支持，不过目前我全身心把空余时间都投入到Wi-Fi开发上了，对于蓝牙的研究只能暂停，目前来看蓝牙兼容性问题并不是这个驱动的问题，而是就像WEG对显卡的修补一个道理，并不能直接在macOS上使用第三方的网卡，因为苹果本身就没有做这方面的兼容。

哈哈 大佬竟然回复我了 你的wifi研究真的太成功了，更新了1.3版本后非常稳定，关于蓝牙方面可能确实会有这些遗憾吧，目前兼容的那几款鼠标太丑了 哈哈哈 所以我打算先买一个usb蓝牙先用着，目前我的 MX anywhere 2s 鼠标可以读到 可以连接 但是在链接后会自动断连 就是这么个情况

不着急的大佬，感觉GitHub真的好和谐，找到了零几年玩贴吧的感觉了 哈哈

khmyznikov, 2021-02-08:
Would be cool if logitech mouse get proper support.

Helloearther, 2021-03-07:
B460主板，AX200，OC0.66，11.2.2，目前用1.3alpha驱动，蓝牙只连airpods，LAN和wifi都连着用，一切正常。

没啥事，就是来感谢下作者大佬。

CPipNoob, 2021-03-12:
> B460主板，AX200，OC0.66，11.2.2，目前用1.3alpha驱动，蓝牙只连airpods，LAN和wifi都连着用，一切正常。
> 
> 没啥事，就是来感谢下作者大佬。

请问IntelBluetoothFirmware是用的1.1.2吗，我使用的airportItlwm-catalina1.2.0。wifi可以使用，但是系统识别不到蓝牙

openviso, 2021-03-18:
> B460主板，AX200，OC0.66，11.2.2，目前用1.3alpha驱动，蓝牙只连airpods，LAN和wifi都连着用，一切正常。
> 
> 没啥事，就是来感谢下作者大佬。
 
你好，蓝牙问题你是怎么解决的呢？

thep0y, 2021-03-19:
> B460主板，AX200，OC0.66，11.2.2，目前用1.3alpha驱动，蓝牙只连airpods，LAN和wifi都连着用，一切正常。
> 
> 没啥事，就是来感谢下作者大佬。

同样是ax200，请问你能正常连接使用蓝牙鼠标吗

openviso, 2021-03-19:
可以正常使用wifi，蓝牙只能连接音箱，不能连接AirPods。也不能使用AirDrop

thep0y, 2021-03-19:
> 可以正常使用wifi，蓝牙只能连接音箱，不能连接AirPods。也不能使用AirDrop

WiFi只能说是能正常上网吧？网速应该达不到满速才对，我用1.3.0的驱动，300M宽带只能跑到100M，1.2.0则只能跑到80M。

Helloearther, 2021-03-20:
> > B460主板，AX200，OC0.66，11.2.2，目前用1.3alpha驱动，蓝牙只连airpods，LAN和wifi都连着用，一切正常。
> > 没啥事，就是来感谢下作者大佬。
> 
> 请问IntelBluetoothFirmware是用的1.1.2吗，我使用的airportItlwm-catalina1.2.0。wifi可以使用，但是系统识别不到蓝牙

都是当前最新版本。我是bigsur

Helloearther, 2021-03-20:
> > B460主板，AX200，OC0.66，11.2.2，目前用1.3alpha驱动，蓝牙只连airpods，LAN和wifi都连着用，一切正常。
> > 没啥事，就是来感谢下作者大佬。
> 
> 同样是ax200，请问你能正常连接使用蓝牙鼠标吗

鼠标没试过，有空试下再看下

Helloearther, 2021-03-20:
> 可以正常使用wifi，蓝牙只能连接音箱，不能连接AirPods。也不能使用AirDrop

我的airpods pro 正常

openviso, 2021-03-20:
> > 可以正常使用wifi，蓝牙只能连接音箱，不能连接AirPods。也不能使用AirDrop
> 
> 我的airpods pro 正常
大佬，你用的哪个蓝牙驱动。方便加一下QQ吗？请教几个问题。

openviso, 2021-03-20:
> > 可以正常使用wifi，蓝牙只能连接音箱，不能连接AirPods。也不能使用AirDrop
> 
> WiFi只能说是能正常上网吧？网速应该达不到满速才对，我用1.3.0的驱动，300M宽带只能跑到100M，1.2.0则只能跑到80M。

我路由器不行，只能跑道100M

closed 2021-03-20 by williambj1