OpenIntelWireless/IntelBluetoothFirmware #253: Bluetooth slows / drops after long usage or engaging mic, potential power issue

ohaikai, 2021-01-25:
**Description**
This has been tested using AirPods primarily, but also with various other bluetooth audio devices and also affects keyboards.

When actively transmitting, the Bluetooth will at some point die. Most apparent when listening to audio, and the audio will slowly cut out until it completely disappears. Reconnecting the device restores the connection. On keyboards, input will start to get delayed until reconnected.

Engaging any sort of Bluetooth mic doesn't work, it also disconnects the audio at the same time. Reconnecting brings the audio back until the next time you try to engage the mic.

Possibly an issue with the Bluetooth being 5.0 on Wifi 6 Chipsets, but the issue might be due to a power offset within the firmware, as the symptoms are indicative of flaky connections.

**Environment**

- Kext Version: 1.1.2
- WiFi Card Model: Intel Wifi 6 B460 Chipset
- Product ID: 0x8087, 0x0029
- macOS Version: OpenCore Catalina 10.15.7

**Info in System Report - Bluetooth**
  Apple Bluetooth Software Version:	7.0.6f8
  Hardware, Features, and Settings:
  Name:	***
  Address:	***
  Bluetooth Low Energy Supported:	Yes
  Handoff Supported:	Yes
  Instant Hot Spot Supported:	Yes
  Manufacturer:	Intel
  Transport:	USB
  Firmware Version:	10.256 (10.256)
  Bluetooth Power:	On
  Discoverable:	Off
  Connectable:	Yes
  Auto Seek Pointing:	Off
  Remote wake:	On
  Vendor ID:	0x8087
  Product ID:	0x0029
  Bluetooth Core Spec:	Unknown (0xA)
  HCI Revision:	0x100
  LMP Version:	Unknown (0xA)
  LMP Subversion:	0x100
  Device Type (Major):	Computer
  Device Type (Complete):	Mac Desktop
  Composite Class Of Device:	0x380104
  Device Class (Major):	0x01
  Device Class (Minor):	0x01
  Service Class:	0x1C0
  Auto Seek Keyboard:	Off
  Devices (Paired, Configured, etc.):
  AirPods Pro:
  Address:	***
  Major Type:	Audio
  Minor Type:	Headphones
  Services:	Audio Sink, AVRCP Controller, AAP Server, Handsfree, AVRCP Target
  Paired:	Yes
  Configured:	Yes
  Connected:	Yes
  Manufacturer:	Apple (0x9, 0x7604)
  Bluetooth Core Spec:	5.0
  Firmware Version:	3​A​283
  Vendor ID:	0x004C
  Product ID:	0x200E
  Class of Device:	0x04 0x06 0x240418
  Role:	Master
  Connection Mode:	Active Mode
  Interval:	0 ms
  EDR Supported:	Yes
  eSCO Supported:	Yes
  SSP Supported:	Yes

**Logs**
N/A

**Kernel Panic Logs**
N/A

**Screenshots**
N/A

williambj1, 2021-01-29:
Interesting thoughts, but unfortunately there is nothing we could do with a firmware uploader. @zxystd is focusing on Wi-Fi right now and doesn't have any plan to do extensive investigation and reverse engineering on Apple Bluetooth drivers. Please share more finding if you discovered, that would be much appreciated.

ohaikai, 2021-02-22:
Opening up a new issue as I have found the issue lies within the firmware / driver and will be opening a new issue

closed 2021-02-22 by ohaikai