OpenIntelWireless/IntelBluetoothFirmware #258: 蓝牙不可用，Kext未加载

Qihuaile, 2021-02-09:
**Description**

如题所示，蓝牙无法正常使用。
在hackintool 系统-杂项中无法看到蓝牙设备
在hackintool 内核扩展-已安装中无法看到IntelBluetoothFirmware和IntelBluetoothInjector
执行log show --last boot | grep IntelFirmware命令后 无任何日志输出

**Environment**
硬件信息
AMD 3950x 
AMD 5500xt
华硕C8I主板

型号：AX200

OpenCore V0.6.3
IntelBluetoothFirmware release-1.1.2

- Kext Version: 1.1.2
- WiFi Card Model: AX200
- Product ID: 
- macOS Version: bigSur 11.0.1

**Info in System Report - Bluetooth**

```

```

**Logs**

```log

```

**Kernel Panic Logs**

```

```

**Screenshots**

williambj1, 2021-02-09:
Mapping USB ports is essential for USB  devices on AMD MBs to work. Closing since this is not the fault of ours.

closed 2021-02-09 by williambj1

Qihuaile, 2021-02-09:
> Mapping USB ports is essential for USB devices on AMD MBs to work. Closing since this is not the fault of ours.

请问可以再描述清楚些么～ 我目前USB设备看起来都可以正常工作的

williambj1, 2021-02-09:
Bluetooth info should be available in sys report. The Bluetooth's USB port should appear in ioreg and Hackintool

Qihuaile, 2021-02-09:
> ioreg

- -! 可以再请教下我需要做些什么吗？ 有文档或者指引也可以。 感激不尽！