OpenIntelWireless/IntelBluetoothFirmware #259: Slow kext loading with lots of Interrupt Wait / continue messages

bolu-atx, 2021-02-15:
**Description**
- Bluetooth/wifi loads fine with no delay when booted into windows and restart into MacOS
- If shutdown directly while in MacOS, upon reboot the OpenCore loading screen will get stuck on Interrupt wait/continue messages
- The bootloader is eventually able to boot, but MacOS will be without Wifi/Bluetooth
- After 5-10 min, wifi/bluetooth resumes back normal operation

**Environment**

- Kext Version: latest
- WiFi Card Model:  0x8087, 0x0aaa
- Product ID: 
- macOS Version: Catalina

**Info in System Report - Bluetooth**
```
  Apple Bluetooth Software Version:	7.0.6f7
  Hardware, Features, and Settings:
  Name:	USSD-OBL-65535
  Address:	04-33-C2-24-80-CD
  Bluetooth Low Energy Supported:	Yes
  Handoff Supported:	Yes
  Instant Hot Spot Supported:	Yes
  Manufacturer:	Intel
  Transport:	USB
  Firmware Version:	8.4352 (8.4352)
  Bluetooth Power:	On
  Discoverable:	Off
  Connectable:	Yes
  Auto Seek Pointing:	Off
  Remote wake:	Off
  Vendor ID:	0x8087
  Product ID:	0x0AA7
  Bluetooth Core Spec:	4.2 (0x8)
  HCI Revision:	0x1100
  LMP Version:	4.2 (0x8)
  LMP Subversion:	0x1100
  Device Type (Major):	Computer
  Device Type (Complete):	Mac Desktop
  Composite Class Of Device:	0x380104
  Device Class (Major):	0x01
  Device Class (Minor):	0x01
  Service Class:	0x1C0
  Auto Seek Keyboard:	Off
  Devices (Paired, Configured, etc.):
  Annie’s AirPods:
  Address:	24-F6-77-EF-F5-3B
  Major Type:	Miscellaneous
  Minor Type:	Unknown
  Services:	
  Paired:	Yes
  Configured:	Yes
  Connected:	No
  Bluetooth Core Spec:	1.0b
  Vendor ID:	0x004C
  Product ID:	0x2002
  Class of Device:	0x00 0x00 0x0000
  Lu, Bo’s Keyboard:
  Address:	80-4A-14-74-DB-10
  Major Type:	Peripheral
  Minor Type:	Keyboard
  Services:	Magic Keyboard with Numeric Keypad
  Paired:	Yes
  Configured:	Yes
  Connected:	No
  Manufacturer:	Broadcom (0x5, 0x240C)
  Bluetooth Core Spec:	3.0
  Firmware Version:	0x0100
  Vendor ID:	0x004C
  Product ID:	0x026C
  Class of Device:	0x05 0x10 0x2540
[... 57 lines cut ...]
2021-02-14 18:09:12.120598-0800 0x1c2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2021-02-14 18:09:22.121605-0800 0x1c2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2021-02-14 18:09:22.121609-0800 0x1c2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2021-02-14 18:09:22.124150-0800 0x1c2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2021-02-14 18:09:32.125158-0800 0x1c2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2021-02-14 18:09:32.125161-0800 0x1c2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2021-02-14 18:09:32.127704-0800 0x1c2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2021-02-14 18:09:42.128713-0800 0x1c2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2021-02-14 18:09:42.128716-0800 0x1c2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2021-02-14 18:09:42.131252-0800 0x1c2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2021-02-14 18:09:52.132261-0800 0x1c2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2021-02-14 18:09:52.132265-0800 0x1c2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2021-02-14 18:09:52.134817-0800 0x1c2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2021-02-14 18:10:02.135827-0800 0x1c2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2021-02-14 18:10:02.135830-0800 0x1c2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2021-02-14 18:10:02.138360-0800 0x1c2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2021-02-14 18:10:12.138396-0800 0x1c2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2021-02-14 18:10:12.138400-0800 0x1c2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2021-02-14 18:10:12.140935-0800 0x1c2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2021-02-14 18:10:22.142349-0800 0x1c2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2021-02-14 18:10:22.142353-0800 0x1c2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2021-02-14 18:10:22.145914-0800 0x1c2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2021-02-14 18:10:32.146937-0800 0x1c2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2021-02-14 18:10:32.146940-0800 0x1c2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
```

bolu-atx, 2021-02-15:
Just an update, I just grepped after I got bluetooth back, here's the messages.

```
2021-02-14 18:15:03.208045-0800 0x1c2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2021-02-14 18:15:03.208049-0800 0x1c2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2021-02-14 18:15:03.210583-0800 0x1c2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2021-02-14 18:15:13.210604-0800 0x1c2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2021-02-14 18:15:13.210608-0800 0x1c2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2021-02-14 18:15:13.213140-0800 0x1c2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2021-02-14 18:15:13.214135-0800 0x1a6      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Intel device is already patched. patch num: 34
2021-02-14 18:15:13.214144-0800 0x1c2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2021-02-14 18:15:13.214148-0800 0x1c2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_EVENT_MASK
2021-02-14 18:15:13.218138-0800 0x1c2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2021-02-14 18:15:13.219125-0800 0x1c2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2021-02-14 18:15:13.219131-0800 0x1c2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: End download
2021-02-14 18:15:13.219133-0800 0x1c2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Clean up...
```

williambj1, 2021-02-25:
We have no plan to optimize this, feel free to contribute and send a pull request, thanks.

closed 2021-02-25 by williambj1