OpenIntelWireless/IntelBluetoothFirmware #261: not work for Intel Wireless-AC 9560 160MHz

omegfa, 2021-02-21:
**Description**
not work , not show anythings .

i used both not work i used 
**Environment**

- Kext Version: 1.1.2
- WiFi Card Model: Intel Wireless-AC 9560 160MHz
- Product ID: 
-(windows) Firmware version : HCI 10.256/LMP 10.256
-(windows) Manufacturer id : 2
- macOS Version: macOS 11.2.1 (20D74)
- addition info : asus strix z390 e - airporttitlwm 1.2.0 kext
**Info in System Report - Bluetooth**

```

```

**Logs**

```log

```

**Kernel Panic Logs**

```
no kernel panic - boot normally
```

**Screenshots**
[image]
[image]

closed 2021-02-21 by williambj1

omegfa, 2021-02-21:
why?

sojab0on, 2021-03-27:
Bluetooth is a separate option in system controls its not a network interface the only thing that can show up in network is an Bluetooth pan connection