OpenIntelWireless/IntelBluetoothFirmware #262: Driver / Firmware fails to switch to lower quality codec when using bluetooth mic like airpods , firmware injection from Windows solves

ohaikai, 2021-02-22:
Description:
I've found that the issue with airpods or any bluetooth mic not working on OpenCore with the OpenIntelWireless project lies within the firmware injector / driver versions within the current version. I'm not too clear as to how to extract drivers and add into this project as there's no documentation, but I imagine the process is close to the BRCMPatch method.

After investigation, when using a bluetooth mic AppleHDA / AppleALC drivers like to switch the codec to a lower quality SOC version, which is why audio quality suffers whenever using a bluetooth mic / headset combo. This is not an issue that can be fixed in OpenCore by swapping audio layouts.

However by booting (via dual-boot) into Windows first where I can:

- Install / Load the most recent Intel AX201 Bluetooth driver / firmware
- Connect bluetooth mics without problems.

After that, rebooting from the Windows OS (not via the power button) into MacOS Big Sur allows for the firmware to keep across reboots, and I am able to use any bluetooth mic / headphone combo such as Airpods successfully without issue.

However hard power reset (power button) and a direct boot into MacOS causes the card to most likely lose the injected firmware and all bluetooth devices begin to fail as soon as you try to engage the mic.

This has been tested using AirPods primarily, but also with various other bluetooth mic/headphone combos.

If one of the repo overseers can update the drivers with the most recent AX201 ones that'd be great, or I'd be happy to do it if I could get a little documentation on how to load new devices / drivers without breaking what's already there.

Environment

    Kext Version: 1.1.2
    WiFi Card Model: Intel Wifi 6 AX 201 z490 Chipset
    Product ID: 0x8087, 0x0029
    macOS Version: OpenCore Big Sur 11.2

Info in System Report - Bluetooth
Apple Bluetooth Software Version: 7.0.6f8
Hardware, Features, and Settings:
Name: ***
Address: ***
Bluetooth Low Energy Supported: Yes
Handoff Supported: Yes
Instant Hot Spot Supported: Yes
Manufacturer: Intel
Transport: USB
Firmware Version: 10.256 (10.256)
Bluetooth Power: On
Discoverable: Off
Connectable: Yes
Auto Seek Pointing: Off
Remote wake: On
Vendor ID: 0x8087
Product ID: 0x0029
Bluetooth Core Spec: Unknown (0xA)
HCI Revision: 0x100
LMP Version: Unknown (0xA)
LMP Subversion: 0x100
Device Type (Major): Computer
Device Type (Complete): Mac Desktop
Composite Class Of Device: 0x380104
Device Class (Major): 0x01
Device Class (Minor): 0x01
Service Class: 0x1C0
Auto Seek Keyboard: Off
Devices (Paired, Configured, etc.):
AirPods Pro:
Address: ***
Major Type: Audio
Minor Type: Headphones
Services: Audio Sink, AVRCP Controller, AAP Server, Handsfree, AVRCP Target
Paired: Yes
Configured: Yes
Connected: Yes
Manufacturer: Apple (0x9, 0x7604)
Bluetooth Core Spec: 5.0
Firmware Version: 3​A​283
Vendor ID: 0x004C
Product ID: 0x200E
Class of Device: 0x04 0x06 0x240418
Role: Master
Connection Mode: Active Mode
Interval: 0 ms
EDR Supported: Yes
eSCO Supported: Yes
SSP Supported: Yes

Logs
N/A

Kernel Panic Logs
N/A

Screenshots
N/A

seungjunProgramming, 2021-02-24:
What do yo mean by booting from windows???
Do you mean clicking reboot from widows and booting into macos?

seungjunProgramming, 2021-02-24:
Oh, yes it does!
My bluetooth used to unpair when I change my mic to my Bluetooth device(airpods and my akg headphone), but rebooting from windows make my mic work!

ohaikai, 2021-02-24:
> Do you mean clicking reboot from widows and booting into macos?

Yes I don't know if it's because of how OpenCore 0.6.6 gets rid of the bootloader, but yes, whether you use BootCamp or not, the most important thing is to reboot from Windows into MacOS through the OS (After you have properly loaded the most recent driver for Windows).

Key is to make sure your machine doesn't do a hard power cycle (turn off turn on) to make sure your firmware injection doesn't get lost.

CrossStone, 2021-03-12:
困扰我很久的问题，一直不知道为什么有时候能用而大部分不能用，按照你说的方法从windows重启后，确实airpods的麦克风就可以用了。。。不知道怎么解决

Sheripov, 2021-03-20:
I have some problem with my laptop. I have ax201. I'll be happy if anybody fix it.

mahdose, 2021-03-20:
To be highlighted

lonacluka, 2021-04-05:
@zxystd 
any chance you can add latest firmware for ax200, or provide some steps on how to do it please? Thanks

RaymdZ, 2021-04-15:
Same problem here with AC9560. 
Hot reboot from Win10 will mostly solve it. However, a sleep & wake cycle will crash it again. 
Hope it could be fixed.

tsihyoung, 2021-04-26:
Did you enable the fast boot feature on BIOS?

ohaikai, 2021-04-26:
> Did you enable the fast boot feature on BIOS?

It is disabled as per the BIOS instructions for Comet Lake / Intel for OpenCore / Dortania guide

tsihyoung, 2021-04-28:
> > Did you enable the fast boot feature on BIOS?
> 
> It is disabled as per the BIOS instructions for Comet Lake / Intel for OpenCore / Dortania guide

Thanks. I can reproduce this 'feature' in my AX200 with MacOS Mojave. But the audio quality is worse than the original firmware in MacOS and similar to that in Windows. When recording using Audition, the output channel is unstable, jumping between 1 and 2. Hope this project can improve further.

AndrewFNelson, 2021-04-29:
I was able to reproduce the same behavior with Ubuntu 20.04 instead of Windows 10.  I disabled the IntelBluetoohFirmware kext (Injector is still there).  Rebooted into Ubuntu 20.04 to get the appropriate Intel 9260 firmware, then simply rebooted into BigSur 11.3 and Bluetooth worked as expected.  Both audio out and input via both a Plantronics Voyager 5200 and a Sony WH-3000XM.  The audio quality changes when the microphone is being used, but this is expected behavior and the same as my MacBook Pro.  In a nutshell, the firmware being sent to the 9260 needs to be updated and I think this fixes the no mic behavior.  For reference I have an i7 8700K on an ASUS ROG Strix B350I running OpenCore 0.6.8 and updated to Big Sur 11.3.

cdevent, 2021-04-29:
Small workaround: 
it's give me a possibility to listen music but mic will not working. 

 if you have a line-in in mic settings  you can select it as default when AirPods are connected. 

 ASUS ROG FX553VD 
 Intel AC7265 
 BigSur 11.2.3

jamesst20, 2021-05-20:
I can confirm on AX1650, rebooting from Windows and booting Mac OS fixes the input problem.

jamesst20, 2021-05-22:
@williambj1 

Just letting you know. I got the ASUS USB BT-400 (Broadcom based) and it suffers from the same audio input issue. System Preference --> Sound slow to open and the input bar is not moving when I speak

[image]

zxystd, 2021-06-21:
Please try the newest commit 247c40b40db11b95ba71d005d09470b003c3553b and the v2.0.0 version https://github.com/OpenIntelWireless/IntelBluetoothFirmware/suites/3039787588/artifacts/69015438

tsihyoung, 2021-06-21:
> Please try the newest commit   ##[247c40b](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/commit/247c40b40db11b95ba71d005d09470b003c3553b) and the v2.0.0 version
> Rewrite the whole driver. Introduce these features:
> 
> 1. Speed up loading and downloading time.
> 2. Fix boot loop issue.
> 3. Fix mic useless problem for AirPods.
> 4. Add support for gen3 devices: ax210 (0x0032, 0x0033).
> 
> If you have different experiences with rebooted from Windows and using IntelBluetoothFirmware.kext, now you can try.
> 
> link: https://github.com/OpenIntelWireless/IntelBluetoothFirmware/suites/3039787588/artifacts/69015438

The mic issue has been solved in the commit ##[247c40b](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/commit/247c40b40db11b95ba71d005d09470b003c3553b) for AX200.

ohaikai, 2021-06-22:
> Please try the newest commit [247c40b](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/commit/247c40b40db11b95ba71d005d09470b003c3553b) and the v2.0.0 version
> Rewrite the whole driver. [...]

@zxystd Confirmed fixed! Overall bluetooth connection is much faster, and handoff somewhat works well now too! Thank you so much for your time in fixing this issue. To the collaborators as well, thank you. I'm sorry I wasn't able to contribute to the project further.

ohaikai, 2021-06-22:
Ah sorry @zxystd and crew I've spoken too soon. While now I can engage the mic without crashing the driver, it causes the bluetooth connection to become unstable after a short amount amount of time. If I do not engage the mic at boot, the connection seems stable for a while, but after some time it still seems to lose connection intermittently.

It seems to be potentially a keepalive or bandwidth issue? When my airpods start losing connection, my bluetooth keyboard also starts to lag.

zxystd, 2021-06-22:
Maybe there is something affected the signal of BT? for example, 2.4ghz network is using or crowned of 2.4ghz wifi surrounded.

jamesst20, 2021-06-22:
@zxystd 

Tested today for about 30 minutes. Appears to be flawless on my Alienware M17 R4 (AX1650)

ohaikai, 2021-06-22:
@zxystd Hmm it seems to have gotten better over time. I don't have a 2.4ghz spectrum in my place, just 5ghz. I am still testing to see if it is potentially just my motherboard or if there is actually a trigger that causes the bluetooth to fail. Hopefully I can close the issue without issues!

ohaikai, 2021-06-29:
@zxystd I think it might be due to heavy CPU processing which just made my application sound start cutting apart as well as lag to the keyboard. I can confirm that through a few restarts (without going into Windows first) that the Bluetooth is now solid and working! Thank you again for your help and work everyone.

closed 2021-06-29 by ohaikai