OpenIntelWireless/IntelBluetoothFirmware #266: [Intel AC-7260] IntelBluetoothFirmware causing "IOUSBHostDevice" kernel msg loop

kushwavez, 2021-03-09:
**Description**
IntelBluetoothFirmware causing `IOUSBHostDevice` kernel msg loop. it's spamming from start to stop.
In the log I can see that when the kext finishes it's initalizing, the msg loop is starting right away.
However there is no problem with the bluetooth itself, it's working okay.
If I don't use the IntelBluetoothFirmware.kext, then there is no msg loop, but of course the bluetooth isn't working then.

**Environment**

- Kext Version: 1.1.2 IntelBluetoothFirmware.kext and IntelBluetoothInjector.kext
- WiFi Card Model: Intel AC-7260
- Product ID: Wifi: 7d0, Bluetooth: 7dc
- macOS Version: Mojave 10.14.6 and Big Sur 11.0.1
- Lenovo T440 (USBs are patched, same thing on Clover or OC)

**Info in System Report - Bluetooth**

```
 Apple Bluetooth szoftververzió:	6.0.14d3
  Hardver, funkciók és beállítások:
  Név:	T440 Pro
  Cím:	5C-C5-D4-98-EC-8D
  Az Energiatakarékos Bluetooth funkció támogatott:	Igen
  A Handoff támogatott:	Nem
  Azonnali hotspot támogatva:	Nem
  Gyártó:	Intel
  Átvitel:	USB
  Firmware verzió:	6.1280 (6.1280)
  Bluetooth tápellátás:	Be
  Felfedezhető:	Ki
  Csatlakoztatható:	Igen
  Mutató automatikus keresése:	Be
  Távoli felébresztés:	Be
  Forgalmazóazonosító:	0x8087
  Termékazonosító:	0x07DC
  HCI verzió:	4.0 (0x6)
  HCI revízió:	0x500
  LMP verzió:	4.0 (0x6)
  LMP alverzió:	0x500
  Eszköztípus (főbb):	Computer
  Eszköztípus (teljes):	Mac Portable
  Eszköz kompozit osztálya:	0x38010C
  Eszközosztály (főbb):	0x01
  Eszközosztály (kisebb):	0x03
  Szolgáltatásosztály:	0x1C0
  Billentyűzet automatikus keresése:	Be
  Eszközök (párosított, konfigurált stb.):
  Jam Replay:
  Cím:	FC-58-FA-D6-BF-69
  Főtípus:	Audio
  Altípus:	Headset
  Szolgáltatások:	Hands-free, Serial Port, Headset, Audio/Video remote Control, Audio Sink
  Párosítva:	Igen
  Konfigurálva:	Igen
  Kapcsolódik:	Nem
  Gyártó:	Cambridge Silicon Radio (0x4, 0x1060)
  Firmware verzió:	0x0000
  Eszköz osztálya:	0x04 0x01 0x260404
  Az EDR támogatott:	Igen
  Az eSCO támogatott:	Nem
  SSP-támogatott:	Igen
  Szolgáltatások:
  Bluetooth fájlátvitel:
  Más eszközökkel látható mappa:	~/Public
  Elemek fogadásakor:	Összes elfogadása figyelmeztetés nélkül
  Állapot:	Kikapcsolva
  Bluetooth fájlcserélő:
  Elfogadott elemek mappája:	~/Downloads
  Más elemek elfogadásakor:	Mentés erre a helyre
  Elemek fogadásakor:	Összes elfogadása figyelmeztetés nélkül
  Állapot:	Kikapcsolva
[... 102 lines cut ...]
000060.675285 IOUSBHostDevice@14500000: IOUSBHostDevice::forcePowerGated: allowing power level to float
000060.675294 IOUSBHostDevice@14500000: IOUSBHostDevice::forcePowerGated: kPowerStateSuspended completed with 0x00000000 after 0ms
000060.675370 IOUSBHostDevice@14500000: IOUSBHostDevice::powerStateWillChangeToGated: kPowerStateOn
000060.675408 IOUSBHostDevice@14500000: IOUSBHostDevice::setPowerStateGated: kPowerStateSuspended -> kPowerStateOn
000060.675427 IOUSBHostDevice@14500000: IOUSBHostDevice::setPowerStateGated: resuming pipes
000060.675494 IOUSBHostDevice@14500000: IOUSBHostDevice::powerChangeDone: 
000060.675528 IOUSBHostDevice@14500000: IOUSBHostDevice::updateIdlePolicyGated: IOUSBHostInterface@0 [<ptr>] has an idle policy of 500ms
000060.675542 IOUSBHostDevice@14500000: IOUSBHostDevice::updateIdlePolicyGated: IOUSBHostInterface@1 [<ptr>] has an idle policy of 500ms
000060.675564 IOUSBHostDevice@14500000: IOUSBHostDevice::updateIdlePolicyGated: consensus idle policy timeout 500ms
000061.232844 IOUSBHostDevice@14500000: IOUSBHostDevice::forcePowerGated: kPowerStateSuspended timeout 1000ms
000061.232915 IOUSBHostDevice@14500000: IOUSBHostDevice::powerStateWillChangeToGated: kPowerStateSuspended
000061.233012 IOUSBHostDevice@14500000: IOUSBHostDevice::setPowerStateGated: kPowerStateOn -> kPowerStateSuspended
000061.233023 IOUSBHostDevice@14500000: IOUSBHostDevice::setPowerStateGated: suspending pipes
000061.233082 IOUSBHostDevice@14500000: IOUSBHostDevice::forcePowerGated: allowing power level to float
000061.233103 IOUSBHostDevice@14500000: IOUSBHostDevice::forcePowerGated: kPowerStateSuspended completed with 0x00000000 after 0ms
000061.233175 IOUSBHostDevice@14500000: IOUSBHostDevice::powerStateWillChangeToGated: kPowerStateOn
000061.233212 IOUSBHostDevice@14500000: IOUSBHostDevice::setPowerStateGated: kPowerStateSuspended -> kPowerStateOn
000061.233235 IOUSBHostDevice@14500000: IOUSBHostDevice::setPowerStateGated: resuming pipes
000061.233303 IOUSBHostDevice@14500000: IOUSBHostDevice::powerChangeDone: 
000061.233315 IOUSBHostDevice@14500000: IOUSBHostDevice::updateIdlePolicyGated: IOUSBHostInterface@0 [<ptr>] has an idle policy of 500ms
000061.233325 IOUSBHostDevice@14500000: IOUSBHostDevice::updateIdlePolicyGated: IOUSBHostInterface@1 [<ptr>] has an idle policy of 500ms
000061.233332 IOUSBHostDevice@14500000: IOUSBHostDevice::updateIdlePolicyGated: consensus idle policy timeout 500ms
[...]
``Log attached:
[bootlog_intelbluetooth.log](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/6108446/bootlog_intelbluetooth.log)

Caipe97, 2021-04-11:
Exactly the same issue with Intel 7265, Thinkpad x250, macOS 10.15 Catalina, paired with itlwm.kext. USB Maps are also patched.

mofand3, 2021-05-05:
remove IntelBluetoothInjector.kext and clear nvram, it works for me. not very sure, I will try more times.

Bughi4, 2021-05-07:
USB Maps patched, Intel AX200, Big Sur 11.3.1, same issue. For me it isn't a loop, it just takes very long to boot (around 5 minutes). It also happens sometimes but I can't really determine when (mostly when I boot from Windows and when I force shut down)

FisherWY, 2021-06-01:
The same issue with Intel 9560, Big Sur 11.4. It took a very long time to boot. The mouse(Logitech G304) will drop frames when it first enter the desktop, and return to normal after 1 minute.

closed 2021-06-22 by zxystd

kushwavez, 2021-06-22:
An explanation about why did you close this would be a good thing

zxystd, 2021-06-22:
@kushwavez Because it is not related to this kext, it is just a firmware uploader.

Bluetooth Device in IORegistry is as an IOUSBHostDevice, for native BT device, it is attached by BroadcomUSBHostController, and using IOBluetoothFamily as an library to communicate with user space's bluetoothd(that's what you see as a bluetooth, you can discover/pair/open/close it).
But for Intel cards, because the drivers what I have mentioned above are designed for Broadcom, so there will have so much incompatible commands/methods, so you can see error log in the kernel log is expected(even I didn't see error info in what you posted here, it is just a common verbose log).