OpenIntelWireless/IntelBluetoothFirmware #271: bluetooth was still working after shutdown the system

EnochYng, 2021-03-24:
**Description**
My computer is NUC8I5, its working very well when I used this bluetooth driver, but I observed an issue that the bluetooth was still working after I shutdown my NUC.

**How did I obtain this conclusion**
I have a bluetooth music box, I can connect it with my NUC and its going well, but I observed the connection couldn't be disconnected after I shutdown My computer (Because there will be a tone when connection disconnect, just like '蓝牙断开'), 
So I suspected its very likely a ISSUE. 

Thanks,
Enoch

EnochYng, 2021-03-24:
Anyway, Huge Thanks for your contributions.

williambj1, 2021-03-24:
Sorry but this has nothing to do with a firmware ***uploader***.

closed 2021-03-24 by williambj1