OpenIntelWireless/IntelBluetoothFirmware #274: 蓝牙连不上

VIPJiang, 2021-04-04:
**Description**

点连接后显示已连接，然后又迅速变成未连接
**Environment**

- Kext Version: 1.1.2
- WiFi Card Model: Intel 8265 AC
- Product ID: 0x0AA7
- macOS Version: 10.13.6

**Info in System Report - Bluetooth**

```
Apple 蓝牙软件版本：	6.0.7f22
  硬件、功能和设置：
  名称：	Simon的MacBook Pro
  地址：	88-B1-11-DF-A3-B2
  支持低功耗蓝牙：	是
  支持“接力”：	是
  支持 Instant Hotspot：	是
  制造商：	Intel
  传输：	USB
  固件版本：	8.4352 (8.4352)
  蓝牙电源：	打开
  可被发现：	打开
  可连接：	是
  自动寻找点：	打开
  远程唤醒：	打开
  供应商 ID：	0x8087
  产品 ID：	0x0AA7
  HCI 版本：	4.2 (0x8)
  HCI 修正版：	0x1100
  LMP 版本：	4.2 (0x8)
  LMP 子版本：	0x1100
  设备类型（主要）：	Computer
  设备类型（完整）：	Mac Portable
  复合类设备：	0x38010C
  设备类（主要）：	0x01
  设备类（次要）：	0x03
  服务类：	0x1C0
  自动寻找键盘：	打开
  设备（已配对、已配置等）：
  FIIL T1 X：
  地址：	B0-F1-A3-25-5E-F3
  主要类型：	Audio
  次要类型：	Headset
  服务：	SerialPort, Hands-Free unit
  已配对：	是
  已配置：	是
  已连接：	否
  制造商：	Unknown (0x9, 0x3232)
  设备类：	0x04 0x01 0x240404
  支持 EDR：	是
  支持 eSCO：	是
  支持 SSP：	是
  Simon的iPhone：
  地址：	DC-0C-5C-D2-42-67
  主要类型：	Phone
  次要类型：	Smartphone
  服务：	Phonebook, GATT, PAN Network Access Profile, AVRCP Device, AVRCP Device, Wireless iAP, Handsfree Gateway, Audio Source, MAP MAS-iOS, Wireless iAP v2
  已配对：	是
  已配置：	是
  已连接：	否
  制造商：	Broadcom (0x8, 0x6103)
  固件版本：	0x0E40
  供应商 ID：	0x004C
  产品 ID：	0x7004
  设备类：	0x02 0x03 0x7A020C
  支持 EDR：	是
  支持 eSCO：	是
  支持 SSP：	是
  服务：
  蓝牙文件传输：
  其他设备可以浏览的文件夹：	~/Public
  当接收项目时：	接受所有文件而不发出警告
  状态：	已停用
  蓝牙文件交换：
  存放已接受项目的文件夹：	~/Downloads
  当接受其他项目时：	存储到位置
  当接收项目时：	接受所有文件而不发出警告
  状态：	已停用
  蓝牙互联网共享：
  状态：	已停用
  传入串口：
  Bluetooth-Incoming-Port：
  RFCOMM 通道：	3
  需要鉴定：	否
  传出串口：
  FIILT1X-SerialPort-2：
  地址：	B0-F1-A3-25-5E-F3
  RFCOMM 通道：	3
  需要鉴定：	否

```

**Logs**

```log
2021-04-05 01:25:41.691030+0800 0x54b      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver init()
2021-04-05 01:25:41.691040+0800 0x54b      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Probe()
2021-04-05 01:25:41.691051+0800 0x54b      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0AA7
2021-04-05 01:25:41.691064+0800 0x54b      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Start()
2021-04-05 01:25:43.191245+0800 0x54b      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: start fail, can not open device
2021-04-05 01:25:43.191259+0800 0x54b      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Clean up...
2021-04-05 01:25:43.192675+0800 0x54b      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Stop()
2021-04-05 01:25:43.192788+0800 0x74       Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver free()
2021-04-05 01:25:43.275406+0800 0x6b5      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver init()
2021-04-05 01:25:43.275415+0800 0x6b5      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Probe()
2021-04-05 01:25:43.275422+0800 0x6b5      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0AA7
2021-04-05 01:25:43.275473+0800 0x6b5      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Start()
2021-04-05 01:25:44.775671+0800 0x6b5      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: set configuration to 1
2021-04-05 01:25:44.776086+0800 0x6b5      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found interface!!!
2021-04-05 01:25:44.776151+0800 0x6b5      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found Interrupt endpoint!
2021-04-05 01:25:44.776447+0800 0x6b5      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found Bulk out endpoint!
2021-04-05 01:25:44.776764+0800 0x6b5      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found Bulk in endpoint!
2021-04-05 01:25:44.777026+0800 0x6b5      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: usb init succeed
2021-04-05 01:25:44.777029+0800 0x6b5      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_RESET
2021-04-05 01:25:44.777218+0800 0x6b5      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2021-04-05 01:25:44.785829+0800 0x6b5      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2021-04-05 01:25:44.785832+0800 0x6b5      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2021-04-05 01:25:44.785987+0800 0x6b5      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2021-04-05 01:25:44.786826+0800 0x12c      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Intel device is already patched. patch num: 14
2021-04-05 01:25:44.786834+0800 0x6b5      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2021-04-05 01:25:44.786837+0800 0x6b5      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_EVENT_MASK
2021-04-05 01:25:44.787010+0800 0x6b5      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2021-04-05 01:25:44.787809+0800 0x6b5      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2021-04-05 01:25:44.787816+0800 0x6b5      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: End download
2021-04-05 01:25:44.787818+0800 0x6b5      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Clean up...```

zxystd, 2021-04-05:
连接什么设备？

williambj1, 2021-04-05:
Duplicate of #51

closed 2021-04-05 by williambj1

VIPJiang, 2021-04-05:
> 连接什么设备？

> 连接什么设备？

蓝牙耳机和蓝牙键盘

VIPJiang, 2021-04-05:
> 连接什么设备？

蓝牙耳机和蓝牙键盘