OpenIntelWireless/IntelBluetoothFirmware #279: Don't working bluetooth

skorpions2000, 2021-04-27:
**Description**
Find devices, try connect, but don't working.

**Environment**

- Kext Version: 1.1.2 - 1.0.3 (try)
- Laptop Model: CHUWI CoreBook X (bt 4.2)
- Product ID: 0x8087, 0x0a2a
- macOS Version: Big Sur

**Info in System Report - Bluetooth**
Don't connect mouse on computer, and any devices.

williambj1, 2021-05-02:
No enough info.

closed 2021-05-02 by williambj1

skorpions2000, 2021-05-25:
> Not enough info.

problem is in driver, problem with buffer. you now about this problem, this problem on Bluetooth 4.x in your driver.

skorpions2000, 2021-05-25:
> Not enough info.

Look, this all information...
https://github.com/OpenIntelWireless/IntelBluetoothFirmware/issues/51

williambj1, 2021-05-25:
Simply look at the code, this driver does not handle anything other than uploading the firmware. Blame Apple.

zxystd, 2021-05-25:
> problem is in driver, problem with buffer. you know about this problem, this problem on Bluetooth 4.x in your driver.

Nothing need to be fix with this driver, Apple didn't native support Intel BT. Off topic.