OpenIntelWireless/IntelBluetoothFirmware #282: Bluetooth hardware not founded

eddie-eternal, 2021-05-05:
**Description**

**Environment**

- Kext Version: 1.1.2
- WiFi Card Model: AX200
- Product ID: 
- macOS Version: 10.14.6

**Info in System Report - Bluetooth**

```

```

**Logs**

```log
2021-05-05 16:28:22.447977+0800 0x3f9      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver init()
2021-05-05 16:28:22.447987+0800 0x3f9      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver Probe()
2021-05-05 16:28:22.447997+0800 0x3f9      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0029
2021-05-05 16:28:22.448007+0800 0x3f9      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver Start()
2021-05-05 16:28:23.948194+0800 0x3f9      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: start fail, can not open device
2021-05-05 16:28:23.948209+0800 0x3f9      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Clean up...
2021-05-05 16:28:23.949134+0800 0x3f9      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver Stop()
2021-05-05 16:28:23.949196+0800 0x74       Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver free()
```

**Kernel Panic Logs**

```

```

**Screenshots**

[image]

williambj1, 2021-05-05:
Map your USB

closed 2021-05-05 by williambj1

eddie-eternal, 2021-05-05:
> ports

would you tell me how to map the USB ports?