OpenIntelWireless/IntelBluetoothFirmware #285: Intel AX1650 - Bootloop

jamesst20, 2021-05-13:
- Kext Version: 1.1.2
- WiFi Card Model: AX1650
- macOS Version: Big Sur 11.3

**Screenshots**
[image]

PKRN0, 2021-05-14:
Having exact same problem with AC 9462, Big Sur 11.3.1.

PKRN0, 2021-05-14:
Just solved it by loading BluetoothFirmware.kext after USBMap kext.

jamesst20, 2021-05-14:
@PKRN0 

Was it working prior 11.3.1? Like 11.2? I haven't tried, the computer was delivered to me 2 days ago, my USB installer is 11.3.1

PKRN0, 2021-05-14:
@jamesst20 I didn't tried any prior version of Big Sur. I succeed to boot hack on my laptop few days ago.. However, I tried Catalina 10.15.7 which I remeber caused the same issue.(Might not correct)

jamesst20, 2021-05-14:
I see. Hopefully @zxystd will have some time to spare for us. I wish I had the knowledge to look into it myself. I'm a develeoper but knows nothing when it comes to driver

PKRN0, 2021-05-14:
> Another thing that could be tried is booting windows and rebooting to Mac os without full shutdown. I think the firmware in this case is supposed to be still injected by Windows

Oh, I didn't know that. If it is also related to Itlwm, I might need more testing.

PKRN0, 2021-05-14:
I need to test it more, but it seems like it boots well with 1.0.3.

logicmoon, 2021-05-15:
Sadly I have the same issue on AX200 Intel card after updating to Big Sur 11.3.1. Except in my case in the boot loop it says "Suspect device firmware: ibt-20-1-3.sfi. In my case eventually it loads the system and Bluetooth is working (however it keeps looping at boot for 2-3 minutes before that).

PKRN0, 2021-05-16:
@logicmoon 
Did you try using ver 1.0.3? Sometimes it just doesn't work, but It is not doing any loops now.

goopy13, 2021-05-17:
Hmm, the first time I installed the kext yesterday night it would loop for me and my wifi/bluetooth would stop working but for some reason (maybe multiple reboots?) today it seems to be working perfectly with no substantial delay/loop when booting.

jamesst20, 2021-05-17:
(Just leaving an update here, I downgraded to Big Sur 11.2.3 and it's flawless). My USB port mapping is properly done unlike on 11.3.1. Can't tell which part solved the issue

williambj1, 2021-05-20:
This is usually caused by macOS having a hard time making the usb available for IntelBluetoothFirmware, injecting the kext with different sequences usually helps.

closed 2021-05-20 by williambj1