OpenIntelWireless/IntelBluetoothFirmware #288: 华硕Z590I--ax210蓝牙冷启动无法使用

wxz8506, 2021-05-21:
华硕Z590I--ax210蓝牙这几天折腾发现，要从win切换到黑苹果，蓝牙才可以用，不知道这个现象跟蓝牙驱动有关系吗，还是其他原因？

zxystd, 2021-05-21:
上日志

wxz8506, 2021-05-21:
输入命令回车之后没有任何反应，是空白的。从win切换到黑苹果，蓝牙可以用，和重启直接进黑苹果，蓝牙不可以用了，输入命令回车之后没有任何反应，是空白的。

zxystd, 2021-05-21:
pid是多少？

mac11的日志系统改变了，要用另外的方式获取日志：
- boot-args添加msgbuf=1048576
- 重启电脑
- sudo dmesg | grep -i "IntelFirmware"

wxz8506, 2021-05-21:
[    2.099469]: IntelFirmware: Driver init()
[    2.099498]: IntelFirmware: Driver Probe()
[    2.099515]: IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0032
[    2.099710]: IntelFirmware: Driver Start()
[    3.600932]: IntelFirmware: set configuration to 1
[    3.601880]: IntelFirmware: Found interface!!!
[    3.601931]: IntelFirmware: Found Interrupt endpoint!
[    3.602374]: IntelFirmware: Found Bulk out endpoint!
[    3.602765]: IntelFirmware: Found Bulk in endpoint!
[    3.603255]: IntelFirmware: usb init succeed
[    3.603273]: IntelFirmware: HCI_OP_INTEL_VERSION
[    3.603669]: IntelFirmware: interrupt wait
[    3.604110]: IntelFirmware: Unsupported Intel hardware platform (0)IntelFirmware: interrupt continue
[    3.604153]: IntelFirmware: End download
[    3.604159]: IntelFirmware: Clean up...

zxystd, 2021-05-21:
三代卡的需要另外处理，等我有时间看看吧

wxz8506, 2021-05-21:
噢噢。好的，谢谢。如果有什么需要配合，可以留言给我。

 

 
&mdash;

closed 2021-06-22 by zxystd