OpenIntelWireless/IntelBluetoothFirmware #292: Intermittent Slow boot - Intel firmware interrupt

madvinegar, 2021-05-24:
Have You Read Our Docs
YES

Are You Reporting A Bug
YES

Environment
macOS Big Sur 11.3.1 dual boot with Win10

Kext Version: AirportItlwm 1.3.0, IntelBluetoothFirmware 1.1.2 , IntelBluetoothInjector 1.1.2
WiFi Card Model: intel dual band Wireless-AC 3168 NGW (stone peak) 8086:2110
Bluetooth: 8087:0AA7 wireless AC 3168 Bluetooth
Product ID: 8086:2110 , 8086:24FB
macOS Version: Big Sur 11.3.1
Description

Slow boot (some times 5 minutes)
The slow boot does not happen every time. But when it happens it gets stuck at the intelfirmware section.
I also attach a photo of the verbose when booting. When I finally get into the desktop, the wifi is not working and I need to switch it off and on again to work. Maybe is the bluetooth to blame...?

Bug Report Archive
[ 502.972368]: IntelFirmware: interrupt continue
[ 502.972374]: IntelFirmware: HCI_OP_INTEL_VERSION
[ 502.975408]: IntelFirmware: interrupt wait
[ 502.976391]: IntelFirmware: Intel device is already patched. patch num: 34IntelFirmware: interrupt continue
[ 502.976404]: IntelFirmware: HCI_OP_INTEL_EVENT_MASK
[ 502.980388]: IntelFirmware: interrupt wait
[ 502.981384]: IntelFirmware: interrupt continue
[ 502.981390]: IntelFirmware: End download
[ 502.981391]: IntelFirmware: Clean up...
[ 502.981631]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[ 502.981642]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[ 502.981650]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[ 502.981656]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[ 502.981663]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[ 502.981669]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[ 502.981675]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[ 502.981681]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[ 502.981687]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[ 502.981692]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[ 502.981697]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[ 502.981703]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[ 502.981709]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[ 502.981715]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[ 502.981720]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[ 502.981726]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[ 502.981732]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[ 502.981737]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[ 502.981743]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[ 502.981749]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[ 502.981754]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[ 502.981759]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[ 502.981765]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[ 502.981776]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[ 502.981781]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[ 502.981788]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[ 502.981792]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[ 502.981797]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[ 502.981802]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[ 502.981807]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[ 502.981813]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[ 502.981818]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[ 502.981826]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[ 502.981835]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[ 502.981840]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[ 502.981847]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[ 502.981854]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[ 502.981860]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[ 502.981865]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[ 502.981872]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[ 502.981880]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[ 502.981886]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[ 502.981891]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[ 502.981897]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[ 502.981904]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[ 502.981910]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[ 502.981915]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[Report_2021-05-19_11-30-59.log](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/6530247/Report_2021-05-19_11-30-59.log)
[Log_2021-05-19_11-34-17.log](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/6530248/Log_2021-05-19_11-34-17.log)
[image]
[image]

kylegorter, 2021-05-26:
One unexplainable fix for my case was to open safari. I’ve tried turning WiFi off and on but that didn’t seem to connect. Only when I open safari and safari would miraculously connect for me when the browser would somehow trigger its “auto refresh” function.

madvinegar, 2021-05-26:
One fix that was suggested by the developer was to use/install only the injector kext (and remove the intelfirmware kext).
Seems to be working. No more slow boots.

kylegorter, 2021-05-26:
But now safari doesn’t help me connect WiFi! Haha.

Nader27, 2021-06-16:
same issue

mifjpn, 2021-06-28:
same issue on BIg Sur 11.4

I try to build these kext myself file,and apply.(ver.2.0.0)

kylegorter, 2021-06-28:
How do you “build the kext yourself”. Is there some sort of better support for your specific model by doing that.

mifjpn, 2021-06-28:
Hello
Use xcode and MacKernelSDK.
Here is yesterday build

https://mifmif.mydns.jp/alpha/?p=1867
Please try transle tab

mifjpn, 2021-06-29:
Hello
I'm back after work.
I stood for more than 8 hours and cold booted, but both Clover and OpenCore start without problems.
There was no indication that slows down the boot, such as "Intel firmware ...".

environment:
          Laptop: NEC Versapro VK23VB-T (OEM of Lenovo T260)
                       Skylake 6th-gen CPU, RAM 16GB, Intel HD Graphics 520 2048 MB

WiFi chip etc:
                      04: 00.0 8086 24F3 8086 1130 L1 
                      Intel Corporation Wireless 8260 Network controller 
                      Network controller 
                      XP3@1C,2/PXSX@0 
                      pci8086,24f3
                      PciRoot(0x0)/Pci(0x1C,0x2)/Pci(0x0,0x0)

Kexts:
          AirportItlwm.kext 1.3.0
          IntelBluetoothFirmware.kext 2.0.0 (beb5224)
          IntelBluetoothInjector.kext 2.0.0 (beb5224)
         (Intel Bluetooth * .kext was built by myself.)

Build environment:
BigSur11.4
        Xcode 12.5.1
    Required SDK:
        Acidanthera / MacKernel SDK

I will try to see if there are any problems after that. 

I'm not native, so please forgive me for terrible English. 

Thank you.
（If you will try Build these kexts,I will help you.)

williambj1, 2021-06-29:
Hi everyone! This issue should be solved with the latest updates. Feel free to continue posting comments if that is not the case.

closed 2021-06-29 by williambj1

kylegorter, 2021-07-02:
Thank you so much for the hard work guys!