OpenIntelWireless/IntelBluetoothFirmware #295: Intel Bluetooth not working in macOS 12 Monterey

AvaQueen-Official, 2021-06-08:
Bluetooth cannot be turned on. I'm using the latest version of both kexts.

ctermiii, 2021-06-08:
me2

AvaQueen-Official, 2021-06-08:
@1Revenger1 so is there any fix yet? Or just hold on while people figure something out?

1Revenger1, 2021-06-08:
no fix yet

AvaQueen-Official, 2021-06-08:
Alright thanks for the quick reply. Do you know the exact cause?

kidonng, 2021-06-09:
Also see acidanthera/bugtracker#1669 (possibly what @1Revenger1 is referring to)

shouri10, 2021-06-13:
There have been developments in BrcmPatchRam for bluetooth to work on Monterey. https://github.com/acidanthera/bugtracker/issues/1669#issuecomment-860125243

hexart, 2021-06-14:
[image]
[image]
BCM1280A已经成功在Monterey下驱动了。

AvaQueen-Official, 2021-06-14:
Nice work @hexart. Does that mean that Intel Bluetooth is on the horizon??? 👀

williambj1, 2021-06-14:
> Nice work @hexart. Does that mean that Intel Bluetooth is on the horizon??? 👀

Maybe not yet, `bluetoothd` still crashes whenever it requests a scan.

pigworlds, 2021-06-15:
Looks like the latest firmware update in commit c52425f1d62eabdc84d502d07b40240b89047978 is not working properly in macOS Monterey. By reverting the firmware to its previous version, it is working on my setup.

To make Intel Bluetooth working in Monterey, following are needed:
- Latest version of Lilu (from source)
- Latest version of BrcmPatchRAM (from source), only need BlueToolFixup.kext in this package
- IntelBluetoothFirmware.kext with older firmware (Can be built from https://github.com/pigworlds/IntelBluetoothFirmware/tree/revertfw)
- IntelBluetoolInjector.kext is not needed.
- Shutdown and restart machine instead of reboot. 

Tested on Wireless-AC 8265 and Wireless-AC 9560 with streaming audio to Bluetooth headset.

1Revenger1, 2021-06-15:
Interesting - works on my AC 9560 as well. Was able to hook up my Magic Trackpad 2 just fine. It still has some issues with turning back on though if you turn it off.

AvaQueen-Official, 2021-06-15:
@pigworlds @1Revenger1 tested and is working on my Intel AX201. Interesting how such an odd configuration works out!

zxystd, 2021-06-15:
Great discovery! It seems that Monterey doesn't support BT 5.1 and BT 5.2, so new firmwares is not supported, caused 718 error code(HCIERROR) error.
I still believe it is a bug about the new macOS version. But we can downgrade the firmwares to BT 5.0 for a temporary solution.

[image]

pigworlds, 2021-06-15:
That's a good point and this sounds like the bug in macOS 12.

On the side, I checked the spec of recent [MacBook Pro (16-inch, 2019)](https://support.apple.com/kb/SP809?locale=en_US), [MacBook Pro (13-inch, M1, 2020)](https://support.apple.com/kb/SP824?locale=en_US), they only support Bluetooth 5.0 wireless technology. There is no point for Apple to add support Bluetooth 5.1 and 5.2 for Intel devices that doesn't exist in the market today.

You can merge my branch if you think it is worth to downgrade the firmware temporarily for now.

zxystd, 2021-06-15:
@pigworlds That's kind of funny, Apple support BT 5.1(unless display in the panel) in Bigsur 11.4, so now in new version OS it rollback to BT 5.0...

pigworlds, 2021-06-15:
How do you know the actual BT version it the firmware supports? What's the version before/after the commit c52425f1d62eabdc84d502d07b40240b89047978?

When I load the downgraded firmware, the system profiler shows 5.1 with Wireless-AC 9560, and 4.2 with Wireless-AC 8265 in Big Sur. This lead me believe the system profiler is just reading the version from the device firmware. This downgraded firmware should support BT 5.1.

zxystd, 2021-06-15:
@pigworlds Initial support for BT gen2 is BT 5.0, and 3a9fd05d23b49a6d02658df83dd3975c18a05bed is BT 5.1 and the last firmware update is BT 5.2.

AvaQueen-Official, 2021-06-15:
Will this become a release @zxystd or should we just build the kexts ourselves? Nice work btw!

pigworlds, 2021-06-15:
> @pigworlds Initial support for BT gen2 is BT 5.0, and [3a9fd05](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/commit/3a9fd05d23b49a6d02658df83dd3975c18a05bed) is BT 5.1 and the last firmware update is BT 5.2.

Thanks for the information!

kidonng, 2021-06-16:
Very interestingly, with @pigworlds's setup I can use my Bluetooth 4.x mouse again. Yeah I'm aware of #51, but actually I have no (serious) issues until macOS 11.4, when I couldn't connect to the mouse at all.

kidonng, 2021-06-17:
Here is another fact I discovered: I updated the firmware to the latest version from Linux (specifically [this commit](https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/commit/intel?id=9be3daa2587e01d6a27e4f8667d3338413884e7f) as I'm using Intel 9461) and it still works. So I guess there could also be a firmware issue.

Can anyone else test if [this build](https://github.com/kidonng/IntelBluetoothFirmware/releases/tag/1.1.3) works?

pigworlds, 2021-06-18:
> Here is another fact I discovered: I updated the firmware to the latest version from Linux (specifically [this [...]

I am unable to make it work with this build on Wireless-AC 9560. Did you shut down the machine between test to make sure the expected firmware is loaded?

kidonng, 2021-06-18:
> I am unable to make it work with this build on Wireless-AC 9560. Did you shut down the machine between test to make sure the expected firmware is loaded?

Yup, my fault. I did reboot, but now I inspected further and found that the system is using the old Firmware kext. Sorry for the disturbance!

closed 2021-06-18 by williambj1

lwndow, 2021-10-05:
> @pigworlds @1Revenger1 tested and is working on my Intel AX201. Interesting how such an odd configuration works out!

This also works with my AX201. BlueTool + IntelFirmware (no injector).

adeelpm, 2021-10-29:
I too have ac 9560,Opencore: 0.7.4,intelBluetoothFirmware: 1.1.3,BlueToolFixup: 2.6.1 with intelbluetoothfirmware first followed by bluetoolfix in kext order but doesn't seems to work and i tried ibf 2.0.1 that didn't work too

jamesst20, 2021-10-29:
> I too have ac 9560,Opencore: 0.7.4,intelBluetoothFirmware: 1.1.3,BlueToolFixup: 2.6.1 with intelbluetoothfirmware first followed by bluetoolfix in kext order but doesn't seems to work and i tried ibf 2.0.1 that didn't work too

Make sure Intel Bluetooth firmware is injected before BluetoolFix. That's what worked for me

sunanan, 2021-10-29:
I tried the new BlueToolFixup 2.6.1 with IntelBluetoothFirmware 2.0.1.
the combo is working.
just only the Trackpad doesn't show the battery percent directly, I need to wait for some minutes.

adeelpm, 2021-10-30:
Thanks guys got it working with new build of bluetoolifxup [8b5c9d7](https://github.com/acidanthera/BrcmPatchRAM/commit/8b5c9d76ad5913d54979c36a8aadf20a08787354) seems like there was some issue with MBP 15,4 board(I didn't know of) that they fixed today

VGerris, 2021-10-30:
hi, do you have a build somewhere I can try? It is not released it seems. thanks

adeelpm, 2021-10-31:
> Latest IntelBluetoothFirmware: https://github.com/OpenIntelWireless/IntelBluetoothFirmware/releases/latest Latest BrcmPatchRAM (BlueToolFixup): https://dortania.github.io/builds/?product=BrcmPatchRAM&viewall=true

@VGerris Use this link

samrth012, 2021-10-31:
No version of IntelBluetooth along with BlueTool (2.6.1) works for me, I have tried all the version from 1.1.1 to 2.0.1 for bluetooth firmware also tried inject firmware before blue tool doesnt work for me. OC 0.7.4, macOS 12.0.1

Please help

samrth012, 2021-10-31:
> > Hi @sunanan can you specify the version of the kext you used..
> 
> Opencore: 0.7.4, intelBluetoothFirmware: 1.1.3, BlueToolFixup: 2.6.1.
this also didnt work for me