OpenIntelWireless/IntelBluetoothFirmware #301: Intermittent Slow boot - Intel firmware interrupt

madvinegar, 2021-06-21:
[Log_2021-05-19_11-34-17.log](https://github.com/OpenIntelWireless/itlwm/files/6507123/Log_2021-05-19_11-34-17.log)
[Report_2021-05-19_11-30-59.log](https://github.com/OpenIntelWireless/itlwm/files/6507124/Report_2021-05-19_11-30-59.log)

**Have You Read Our Docs**
YES

**Are You Reporting A Bug**
YES

**Environment**
macOS Big Sur 11.3.1 dual boot with Win10

- Kext Version: AirportItlwm 1.3.0, IntelBluetoothFirmware 1.1.2 , IntelBluetoothInjector 1.1.2
- WiFi Card Model: intel dual band Wireless-AC 3168 NGW (stone peak) 8086:2110
- Bluetooth: 8087:0AA7 wireless AC 3168 Bluetooth
- Product ID: 8086:2110 , 8086:24FB
- macOS Version: Big Sur 11.3.1

**Description**

Slow boot (some times 5 minutes)
The slow boot does not happen every time. But when it happens it gets stuck at the intelfirmware section.
I also attach a photo of the verbose when booting. When I finally get into the desktop, the wifi is not working and I need to switch it off and on again to work. Maybe is the bluetooth to blame...?

**Bug Report Archive**
[  502.972368]: IntelFirmware: interrupt continue
[  502.972374]: IntelFirmware: HCI_OP_INTEL_VERSION
[  502.975408]: IntelFirmware: interrupt wait
[  502.976391]: IntelFirmware: Intel device is already patched. patch num: 34IntelFirmware: interrupt continue
[  502.976404]: IntelFirmware: HCI_OP_INTEL_EVENT_MASK
[  502.980388]: IntelFirmware: interrupt wait
[  502.981384]: IntelFirmware: interrupt continue
[  502.981390]: IntelFirmware: End download
[  502.981391]: IntelFirmware: Clean up...
[  502.981631]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981642]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981650]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981656]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981663]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981669]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981675]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981681]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981687]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981692]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981697]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981703]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981709]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981715]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981720]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981726]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981732]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981737]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981743]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981749]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981754]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981759]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981765]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981776]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981781]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981788]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981792]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981797]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981802]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981807]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981813]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981818]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981826]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981835]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981840]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981847]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981854]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981860]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981865]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981872]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981880]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981886]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981891]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981897]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981904]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981910]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981915]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)

**Kext Download Source**
Downloaded from official GitHub page

[image]
[image]

madvinegar, 2021-06-21:
> @madvinegar Please try the latest v2.0.0 version of IntelBluetoothFirmware.kext. Anything update please go to that repo. I am going to close this issue, thanks.

Do I need both kexts? I have removed the intelfirmware kext, I only use the injector kexts, and everything is working fine.
What will it be improved if I install both kexts?

williambj1, 2021-06-22:
Without the firmware uploader, your Bluetooth will not work if you performed a cold boot. Please read the docs for more info.

madvinegar, 2021-06-22:
> Without the firmware uploader, your Bluetooth will not work if you performed a cold boot. Please read the docs for more info.

Thank you for the clarification. Where can I find latest v2.0.0 version?

williambj1, 2021-06-22:
> Where can I find the v2.0.0 kexts?

Located in GitHub Actions' Artifacts section.