OpenIntelWireless/IntelBluetoothFirmware #303: 7265 AC kernel panic on 2.0.0

SkyrilHD, 2021-06-24:
**Description**
Sometimes IntelBluetooth is causing kernel panic.

**Environment**

- Kext Version: 2.0.0
- WiFi Card Model: Intel Dual Band Wireless-AC 7265
- Product ID: 0x8086 0x095A
- macOS Version: macOS 11.3.1 Big Sur

**Info in System Report - Bluetooth**

```
Apple Bluetooth-Softwareversion:	8.0.4d18
  Hardware, Funktionen und Einstellungen:
  Name:	MacBook Pro von Berk
  Adresse:	4C-34-88-DF-C0-FE
  Bluetooth Low Energy wird unterstützt:	Ja
  Handoff wird unterstützt:	Ja
  Instant Hotspot unterstützt:	Ja
  Hersteller:	Intel
  Transport:	USB
  Firmwareversion:	8.4096 (8.4096)
  Bluetooth-Signal:	Ein
  Sichtbar:	Aus
  Verfügbar:	Ja
  Zeigegerät automatisch suchen:	Ein
  Entferntes Beenden des Ruhezustands:	Ein
  Hersteller-ID:	0x8087
  Produkt-ID:	0x0A2A
  Bluetooth-Kernspezifikation:	4.2 (0x8)
  HCI-Revision:	0x1000
  LMP-Version:	4.2 (0x8)
  LMP-Unterversion:	0x1000
  Gerätetyp (Obergruppe):	Computer
  Gerätetyp (komplett):	Mac Portable
  Composite-Klasse des Geräts:	0x38010C
  Geräteklasse (Obergruppe):	0x01
  Geräteklasse (Untergruppe):	0x03
  Dienstklasse:	0x1C0
  Tastatur automatisch suchen:	Ein
```

**Logs**

Terminal does not output anything

**Kernel Panic Logs**

```
panic(cpu 1 caller 0xffffff8019dc52a6): Kernel trap at 0xffffff801cab3346, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0x00000000e000041b, CR3: 0x000000002548b000, CR4: 0x00000000003626e0
RAX: 0x0000000000000000, RBX: 0x0000000000000026, RCX: 0x0000000000000000, RDX: 0x0000000000000001
RSP: 0xffffffa081b8bb70, RBP: 0xffffffa081b8bb80, RSI: 0x0000000000000001, RDI: 0xffffff936f9cc7b0
R8:  0x000000000000ffff, R9:  0x00000000ffffffff, R10: 0x0000000000000001, R11: 0xffffff86a3807000
R12: 0x00000000e00002cd, R13: 0x0000000000000000, R14: 0x00000000e00002eb, R15: 0x0000000000000000
RFL: 0x0000000000010283, RIP: 0xffffff801cab3346, CS:  0x0000000000000008, SS:  0x0000000000000000
Fault CR2: 0x00000000e000041b, Error code: 0x0000000000000000, Fault CPU: 0x1, PL: 0, VF: 1

Backtrace (CPU 1), Frame : Return Address
0xffffffa081b8b590 : 0xffffff8019c8e02d 
0xffffffa081b8b5e0 : 0xffffff8019dd48e3 
0xffffffa081b8b620 : 0xffffff8019dc4eda 
0xffffffa081b8b670 : 0xffffff8019c32a2f 
0xffffffa081b8b690 : 0xffffff8019c8d84d 
0xffffffa081b8b7b0 : 0xffffff8019c8db43 
0xffffffa081b8b820 : 0xffffff801a49d68a 
0xffffffa081b8b890 : 0xffffff8019dc52a6 
0xffffffa081b8ba10 : 0xffffff8019dc4f8d 
0xffffffa081b8ba60 : 0xffffff8019c32a2f 
0xffffffa081b8ba80 : 0xffffff801cab3346 
0xffffffa081b8bb80 : 0xffffff801a3f17bf 
0xffffffa081b8bbe0 : 0xffffff801cab250b 
0xffffffa081b8bc50 : 0xffffff801a3f17bf 
0xffffffa081b8bcb0 : 0xffffff801cab2135 
0xffffffa081b8bd00 : 0xffffff801ea82dd5 
0xffffffa081b8bd20 : 0xffffff801ea84ca2 
0xffffffa081b8bd40 : 0xffffff801ea86cd0 
0xffffffa081b8bd60 : 0xffffff801ea86c2c 
0xffffffa081b8bdd0 : 0xffffff801a3c69ab 
0xffffffa081b8be40 : 0xffffff801a3c653e 
0xffffffa081b8bef0 : 0xffffff801a3c56df 
0xffffffa081b8bf50 : 0xffffff801a3c8456 
0xffffffa081b8bfa0 : 0xffffff8019c3213e 
      Kernel Extensions in backtrace:
         com.apple.iokit.IOUSBHostFamily(1.2)[5190CA74-773E-3915-B4AA-8FDB4D1ED52B]@0xffffff801ca7c000->0xffffff801cb0afff
            dependency: com.apple.driver.AppleBusPowerController(1.0)[66C01074-5C99-3222-91D1-F80B98D55EDD]@0xffffff801b02e000->0xffffff801b031fff
            dependency: com.apple.driver.AppleSMC(3.1.9)[406AA67A-5111-3CA1-8C7A-A95B18AF6252]@0xffffff801b34e000->0xffffff801b366fff
            dependency: com.apple.driver.usb.AppleUSBCommon(1.0)[96B84822-9BC5-3DC8-86F0-6E38178E615B]@0xffffff801b49e000->0xffffff801b4a1fff
            dependency: com.apple.driver.AppleUSBHostMergeProperties(1.2)[CFEDF52B-3A3F-30DF-B042-81434B369E0F]@0xffffff801cb77000->0xffffff801cb77fff
            dependency: com.apple.iokit.IOACPIFamily(1.4)[0D13DDA8-F6FE-3649-A850-DE1D9BFF8CEF]@0xffffff801c23a000->0xffffff801c23bfff
         com.zxystd.IntelBluetoothFirmware(2.0)[179C1964-1D27-35E3-ABF3-D555FF925034]@0xffffff801ea82000->0xffffff801ed37fff
            dependency: com.apple.iokit.IOUSBHostFamily(1.2)[5190CA74-773E-3915-B4AA-8FDB4D1ED52B]@0xffffff801ca7c000->0xffffff801cb0afff

Process name corresponding to current thread: kernel_task
Boot args: itlwm_cc=EU -lilubetaall -vsmcbeta chunklist-security-epoch=0 -chunklist-no-rev2-dev chunklist-security-epoch=0 -chunklist-no-rev2-dev

Mac OS version:
20E241

Kernel version:
Darwin Kernel Version 20.4.0: Thu Apr 22 21:46:47 PDT 2021; root:xnu-7195.101.2~1/RELEASE_X86_64
Kernel UUID: 068F99A3-1DB3-31C0-87D5-09942F122BB6
KernelCache slide: 0x0000000019a00000
KernelCache base:  0xffffff8019c00000
Kernel slide:      0x0000000019a10000
Kernel text base:  0xffffff8019c10000
__HIB  text base: 0xffffff8019b00000
System model name: MacBookPro12,1 (Mac-E43C1C25D4880AD6)
System shutdown begun: NO
Panic diags file available: NO (0xe00002d8)
Hibernation exit count: 0

System uptime in nanoseconds: 3789569532
Last Sleep:           absolute           base_tsc          base_nano
  Uptime  : 0x00000000e1e04673
  Sleep   : 0x0000000000000000 0x0000000000000000 0x0000000000000000
  Wake    : 0x0000000000000000 0x00000010f15fa8b0 0x0000000000000000
```

**Additional Info**

I also wanted to mention that when I try booting to Monterey, it gets stuck in verbose and does not boot. I have to restart the whole system and try booting again (which works most of the cases). I don't know how to extract verbose logs, so I cannot provide them here.

zxystd, 2021-06-24:
please add keepsyms=1 in boot-args, and can you tell me which kext you are using corresponding to which commit?

SkyrilHD, 2021-06-24:
Oops forgot to add the boot-arg, I rebooted like 5-6 times and I couldn't cause the kernel panic. I will post the log, if it panics.

It is this commit: https://github.com/OpenIntelWireless/IntelBluetoothFirmware/commit/106cd8a19128a539921563a320279f903f891264
I downloaded it from the Artifacts

SkyrilHD, 2021-06-24:
Here is the log... It happens if I shutdown the system for 10 minutes and then trying to boot it back up.

```
panic(cpu 3 caller 0xffffff80127c52a6): Kernel trap at 0xffffff80154b3346, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0x00000000e000041b, CR3: 0x000000001de8b000, CR4: 0x00000000003626e0
RAX: 0x0000000000000000, RBX: 0x0000000000000026, RCX: 0x0000000000000000, RDX: 0x0000000000000001
RSP: 0xffffffa0a5493b70, RBP: 0xffffffa0a5493b80, RSI: 0x0000000000000001, RDI: 0xffffff936860ab10
R8:  0x000000000000ffff, R9:  0x00000000ffffffff, R10: 0x0000000000000001, R11: 0x0000000000000000
R12: 0x00000000e00002cd, R13: 0x0000000000000000, R14: 0x00000000e00002eb, R15: 0x0000000000000000
RFL: 0x0000000000010283, RIP: 0xffffff80154b3346, CS:  0x0000000000000008, SS:  0x0000000000000010
Fault CR2: 0x00000000e000041b, Error code: 0x0000000000000000, Fault CPU: 0x3, PL: 0, VF: 1

Backtrace (CPU 3), Frame : Return Address
0xffffffa0a5493590 : 0xffffff801268e02d mach_kernel : _handle_debugger_trap + 0x3fd
0xffffffa0a54935e0 : 0xffffff80127d48e3 mach_kernel : _kdp_i386_trap + 0x143
0xffffffa0a5493620 : 0xffffff80127c4eda mach_kernel : _kernel_trap + 0x55a
0xffffffa0a5493670 : 0xffffff8012632a2f mach_kernel : _return_from_trap + 0xff
0xffffffa0a5493690 : 0xffffff801268d84d mach_kernel : _DebuggerTrapWithState + 0xad
0xffffffa0a54937b0 : 0xffffff801268db43 mach_kernel : _panic_trap_to_debugger + 0x273
0xffffffa0a5493820 : 0xffffff8012e9d68a mach_kernel : _panic + 0x54
0xffffffa0a5493890 : 0xffffff80127c52a6 mach_kernel : _sync_iss_to_iks + 0x2c6
0xffffffa0a5493a10 : 0xffffff80127c4f8d mach_kernel : _kernel_trap + 0x60d
0xffffffa0a5493a60 : 0xffffff8012632a2f mach_kernel : _return_from_trap + 0xff
0xffffffa0a5493a80 : 0xffffff80154b3346 com.apple.iokit.IOUSBHostFamily : __ZN18IOUSBHostInterface15closePipesGatedEv + 0x18
0xffffffa0a5493b80 : 0xffffff8012df17bf mach_kernel : __ZN13IOCommandGate9runActionEPFiP8OSObjectPvS2_S2_S2_ES2_S2_S2_S2_ + 0xaf
0xffffffa0a5493be0 : 0xffffff80154b250b com.apple.iokit.IOUSBHostFamily : __ZN18IOUSBHostInterface10closeGatedEP9IOServicej + 0x31
0xffffffa0a5493c50 : 0xffffff8012df17bf mach_kernel : __ZN13IOCommandGate9runActionEPFiP8OSObjectPvS2_S2_S2_ES2_S2_S2_S2_ + 0xaf
0xffffffa0a5493cb0 : 0xffffff80154b2135 com.apple.iokit.IOUSBHostFamily : __ZN18IOUSBHostInterface5closeEP9IOServicej + 0x6f
0xffffffa0a5493d00 : 0xffffff8017482dd5 com.zxystd.IntelBluetoothFirmware : __ZN19USBDeviceController4freeEv + 0x117
0xffffffa0a5493d20 : 0xffffff8017484ca2 com.zxystd.IntelBluetoothFirmware : __ZN7BtIntel4freeEv + 0x34
0xffffffa0a5493d40 : 0xffffff8017486cd0 com.zxystd.IntelBluetoothFirmware : __ZN22IntelBluetoothFirmware7cleanUpEv + 0x30
0xffffffa0a5493d60 : 0xffffff8017486c2c com.zxystd.IntelBluetoothFirmware : __ZN22IntelBluetoothFirmware5startEP9IOService + 0x210
0xffffffa0a5493dd0 : 0xffffff8012dc69ab mach_kernel : __ZN9IOService14startCandidateEPS_ + 0xeb
0xffffffa0a5493e40 : 0xffffff8012dc653e mach_kernel : __ZN9IOService15probeCandidatesEP12OSOrderedSet + 0xcce
0xffffffa0a5493ef0 : 0xffffff8012dc56df mach_kernel : __ZN9IOService14doServiceMatchEj + 0x30f
0xffffffa0a5493f50 : 0xffffff8012dc8456 mach_kernel : __ZN15_IOConfigThread4mainEPvi + 0x186
0xffffffa0a5493fa0 : 0xffffff801263213e mach_kernel : _call_continuation + 0x2e
      Kernel Extensions in backtrace:
         com.apple.iokit.IOUSBHostFamily(1.2)[5190CA74-773E-3915-B4AA-8FDB4D1ED52B]@0xffffff801547c000->0xffffff801550afff
            dependency: com.apple.driver.AppleBusPowerController(1.0)[66C01074-5C99-3222-91D1-F80B98D55EDD]@0xffffff8013a2e000->0xffffff8013a31fff
            dependency: com.apple.driver.AppleSMC(3.1.9)[406AA67A-5111-3CA1-8C7A-A95B18AF6252]@0xffffff8013d4e000->0xffffff8013d66fff
            dependency: com.apple.driver.usb.AppleUSBCommon(1.0)[96B84822-9BC5-3DC8-86F0-6E38178E615B]@0xffffff8013e9e000->0xffffff8013ea1fff
            dependency: com.apple.driver.AppleUSBHostMergeProperties(1.2)[CFEDF52B-3A3F-30DF-B042-81434B369E0F]@0xffffff8015577000->0xffffff8015577fff
            dependency: com.apple.iokit.IOACPIFamily(1.4)[0D13DDA8-F6FE-3649-A850-DE1D9BFF8CEF]@0xffffff8014c3a000->0xffffff8014c3bfff
         com.zxystd.IntelBluetoothFirmware(2.0)[179C1964-1D27-35E3-ABF3-D555FF925034]@0xffffff8017482000->0xffffff8017737fff
            dependency: com.apple.iokit.IOUSBHostFamily(1.2)[5190CA74-773E-3915-B4AA-8FDB4D1ED52B]@0xffffff801547c000->0xffffff801550afff

Process name corresponding to current thread: kernel_task
Boot args: itlwm_cc=EU -lilubetaall -vsmcbeta -v keepsyms=1 chunklist-security-epoch=0 -chunklist-no-rev2-dev

Mac OS version:
20E241

Kernel version:
Darwin Kernel Version 20.4.0: Thu Apr 22 21:46:47 PDT 2021; root:xnu-7195.101.2~1/RELEASE_X86_64
Kernel UUID: 068F99A3-1DB3-31C0-87D5-09942F122BB6
KernelCache slide: 0x0000000012400000
KernelCache base:  0xffffff8012600000
Kernel slide:      0x0000000012410000
Kernel text base:  0xffffff8012610000
__HIB  text base: 0xffffff8012500000
System model name: MacBookPro12,1 (Mac-E43C1C25D4880AD6)
System shutdown begun: NO
Panic diags file available: NO (0xe00002d8)
Hibernation exit count: 0

System uptime in nanoseconds: 6925280922
Last Sleep:           absolute           base_tsc          base_nano
  Uptime  : 0x000000019e30e1a7
  Sleep   : 0x0000000000000000 0x0000000000000000 0x0000000000000000
  Wake    : 0x0000000000000000 0x0000000ee6c8f9b1 0x0000000000000000
```

williambj1, 2021-06-24:
> I also wanted to mention that when I try booting to Monterey, it gets stuck in verbose and does not boot.

Could you please updated a picture? Thanks!

SkyrilHD, 2021-06-24:
Sure, let me shutdown the system for another 10 minutes

zxystd, 2021-06-24:
I am on my way home and will post a fixed kext after a while, please help me to test.

SkyrilHD, 2021-06-24:
@williambj1
It's stuck here.
[image]

And here is a picture with IntelBluetooth mentioned:
[image]

My guess is, it's probably crashing as Big Sur but does not panic for some reason

zxystd, 2021-06-24:
[IntelBluetoothFirmware.kext.zip](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/6709845/IntelBluetoothFirmware.kext.zip)
please try this one @SkyrilHD .

SkyrilHD, 2021-06-24:
Ok, first boot did not panic... But I will try 2-3 times and report back

SkyrilHD, 2021-06-24:
So, I booted four more times (including Monterey) and no kernel panics so far! Great job!

SkyrilHD, 2021-06-24:
Will do! But right now it is working just fine

SkyrilHD, 2021-06-24:
@zxystd But I have a question though. There are already issues about it, but is there any way that I can help you with Bluetooth range? Because when I connect my AirPods, the range is really low until I turn off Bluetooth.

zxystd, 2021-06-29:
@SkyrilHD 7265 have BT Coex problems with WiFi, it is a hardware related problems, Intel writes Coex code in iwlwifi while our itlwm don't have. I don't have this card and not plan to do this in short, because it needs lots of test and will affect other cards maybe.

closed 2021-06-29 by zxystd