OpenIntelWireless/IntelBluetoothFirmware #304: v2.0.0 RC1 Black Screen on boot

megamaxs1234, 2021-06-29:
Description
when start up my computer and open macOS Black Screen is shown on boot. if I restart the computer it opens macOS Big Sur without problems, Note that this problem occurs only after updating kext to IntelBluetoothFirmware v2.0.0 RC1 , in the older version of kext v1.1.3  I was having an infinite loop on boot.

Environment
macOS Big Sur 11.4 dual boot with Win10

Kext Version: itlwm 1.3.0 ,AirportItlwm 1.3.0, IntelBluetoothFirmware v2.0.0 RC1 , IntelBluetoothInjector v2.0.0 RC1
WiFi Card Model: Wi-Fi 6 AX201
Product ID: 0x0026
macOS Version: macOS Big Sur 11.4

zxystd, 2021-06-29:
Hackintosh related.

closed 2021-06-29 by zxystd

williambj1, 2021-06-29:
Keep in mind that IntelBluetoothInjector is a plist-only kext.