OpenIntelWireless/IntelBluetoothFirmware #307: Please add back to KEXT (in repo) the correct version/module info for Hackintool to detect

mackonsti, 2021-07-03:
**Description**
Something was changed in the recent code in your repository and Hackintool cannot read the latest version information (beyond the installed stable v1.1.3 of this kext).

**Environment**

- Kext Version: IntelBluetoothFirmware v1.1.3 (installed)
- WiFi Card Model: Intel Wireless-AC 9560
- Product ID: Intel Bluetooth 5.1 [8087:0aaa]
- macOS Version: Catalina 10.15.7

**Info in System Report - Bluetooth**
No need, in my opinion, as it seems a repository issue but you are welcome to ask me or check Hackintool by yourselves.

**Logs**
No need, in my opinion, as it seems a repository issue but you are welcome to ask me or check Hackintool by yourselves.

**Kernel Panic Logs**
No need, in my opinion, as it seems a repository issue but you are welcome to ask me or check Hackintool by yourselves.

**Screenshots**
[image]

closed 2021-07-03 by williambj1

williambj1, 2021-07-03:
Hi @mackonsti 

I believe there is a bug in Hackintool and this should be reported there instead. As long as the version could be interpreted correctly by Xcode our project, it is a valid configuration. However, I also removed the redundant `CURRENT_PROJECT_VERSION` just to make things easier.

mackonsti, 2021-07-03:
Thank you @williambj1 apologies for the "false" alarm and for copying me in the ticket raised to its developer.

commit b8646803 (2021-07-03): Remove redundant `CURRENT_PROJECT_VERSION`

Fix #307