OpenIntelWireless/IntelBluetoothFirmware #308: Not working on Mac OS Monterey dev beta 1

likkai, 2021-07-03:
I've tried from version 1.1.2 to 2.0.0 RC but all of them doesn't work on my device. Please help me solve this. 
[image]

williambj1, 2021-07-03:
Not enough info & invalid issue format. Read the docs first.

closed 2021-07-03 by williambj1

wangzikang-debug, 2021-07-04:
12系统已经可以使用了，蓝牙鼠标也正常使用了

likkai, 2021-07-04:
My bad, here is the required info

**Description**
The 'turn on bluetooth' button became unresponsive after I clicked it and the bluetooth never turns on. 
https://user-images.githubusercontent.com/76016664/124373192-be55eb00-dcc2-11eb-87ea-e2fdb32563d0.mov

**Environment**
Kext version: 1.1.2 to 2.0.0 RC
Wifi card model: Intel Wireless-AC 9560 160MHz
Product ID: This is what I got from running`lspci -knn | grep Net -A2; lsusb`
8086:a370 or 8086:0034
[image]
macOS version: 12.0 developer beta 1
Bootloader: OpenCore 0.7.0

**Info in System Report - Bluetooth**
Bluetooth Controller:
 Address:	NULL
 State:	Off
 Discoverable:	Off
Firmware Version:	v0 c0
 Product ID:	0x0001
 Supported services:	0x381939 < HFP AVRCP A2DP HID NtSv Game NtCl AACP GATT Serial >
 Transport:	UART
 Vendor ID:	0x004C (Apple)

**Logs**
No output from command `log show --last boot | grep IntelFir`
Kext load successfully:`   116    0 0                  0x9d2000   0x9d2000   com.zxystd.IntelBluetoothFirmware (1.1.3) EED47901-F352-3C8B-8361-572CBA2B4D00 <24 7 6 3>`

**Kernel Panic Logs**
None

**Screenshots**
[image]

likkai, 2021-07-04:
> 12系统已经可以使用了，蓝牙鼠标也正常使用了

Could it be the problem of my kext conflicting with this?
有可能是我其他的 kext 影响到这个吗？我还是用不了蓝牙

This is my kexts:
AirportItlwm
AppleALC
BrightnessKeys
CPUFriend
CPUFriendDataProvider
CpuTscSync
CtInaAHCIPort
ECEnabler
IntelBluetoothFirmware
IntelBluetoothInjector
Lilu
NVMeFix
RealtekRTL8111
SMCBatteryManager
SMCProcessor
SMCSuperIO
USBMap
VerbStub
VirtualSMC
VoodooI2C
VoodooI2CHID
VoodooPS2Controller
WhateverGreen

zxystd, 2021-07-04:
@likkai  If you are using mac12, please replace `IntelBluetoothInjector .kext` with `BlueToolFixup.kext` from https://github.com/acidanthera/BrcmPatchRAM

I hope you can read our docs or release notes carefully.

likkai, 2021-07-05:
Sorry, my bad. Thanks for the help.

natedoggfdfl, 2021-07-28:
Hey @likkai Im having the same issue but have no idea what to do. Could you give me some direction on how you were able to resolve this?

likkai, 2021-07-29:
> Hey @likkai Im having the same issue but have no idea what to do. Could you give me some direction on how you were able to resolve this?

Use `IntelBluetoothFirmware.kext` and `BlueToolFixup.kext` from https://github.com/acidanthera/BrcmPatchRAM

yatak15, 2021-11-08:
Hi @ikkai. I upgraded macOS Monterey and same issue on my laptop. 

as you wrote 
Use IntelBluetoothFirmware.kext and BlueToolFixup.kext from https://github.com/acidanthera/BrcmPatchRAM
I couldn't find "IntelBluetoothFirmware.kext " and not sure how to use "BlueToolFixup.kext"...

I'm not programmer just search from duckduckgo. I couldn't understand how to install, update .kext files..
please let me support.

[image]

likkai, 2021-11-08:
> Hi @ikkai. I upgraded macOS Monterey and same issue on my laptop.
>
> as you wrote Use IntelBluetoothFirmware.kext and BlueToolFixup.kext from https://github.com/acidanthera/BrcmPatchRAM I couldn't [...]

Diwnload the zip file released in BrcmPatchRAM. Both files are inside the zip. Then move both of the .kext files into your EFI/OC/Kexts

yatak15, 2021-11-08:
Thank you. @likkai  

I downloaded it as follow. BrcmPatchRAM-master
[image]

but still couldn't find IntelBluetoothFirmware.kext, and "BlueToolFixup.kext" as on README.md. 
if anything wrong place as download location?