OpenIntelWireless/IntelBluetoothFirmware #309: BTLE won't

mackonsti, 2021-07-03:
**Description**

**Environment**

- Kext Version: 
- WiFi Card Model: 
- Product ID: 
- macOS Version: 

**Info in System Report - Bluetooth**

```

```

**Logs**

```log

```

**Kernel Panic Logs**

```

```

**Screenshots**

wangzikang-debug, 2021-07-04:
你可以用构建版2.0，为何执着1.3呢

zxystd, 2021-07-04:
I have ac9260 and it works fine, the same as yours according to the screenshot.

closed 2021-07-04 by zxystd

mackonsti, 2021-07-04:
Hello @zxystd thank you but in my case, with BigSur 11.4.0 there is no possibility to turn ON or OFF the device.
I have tried changing the order of the kext in the OpenCore configuration but without results.
Please, can you provide here and kindly update your instructions on the **order** of the kexts that need to load?

IntelBluetoothFirmware.kext **first?**
IntelBluetoothInjector.kext **second?**
Does the presence or order of **AirportItlwm.kext** have any impact?

Referring to OpenCore 0.7.0 and BigSur 11.4.0 please. In both my machines (just installed on my Lenovo laptop the 9260 NFF card) I get the exact same behaviour with the ON/OFF toggle...

Thank you for your time and patience!

zxystd, 2021-07-04:
@mackonsti AFAIK, there is no difference with the injection order, so I can't figure out what is the 'correct order'.
Have you tried disable `IntelBluetoothInjector.kext`?

mackonsti, 2021-07-04:
Hi @zxystd thanks for your time. So I tried the "Reset Module" hidden menu (Shift+Option+click on icon) to no avail.

I have played with the order of kexts in OpenCore and placed USBPorts.kext **before** any BT kexts [according to this post here](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/issues/170#issuecomment-798907449) but still to no avail. My correct USB port is mapped as internal and in **IORegistryExplorer** I do see `IntelBluetoothFirmware` being attached under it.

Then I added `msgbuf=1048576` in my boot-args of OpenCore, to get more debugging info but `log show --last boot | grep IntelFir` doesn't produce results, still.

What OpenCore and BigSur are **you** using, please? And what would you advise for the issue where no debug results at all appear?

**UPDATE:** It appears that with `msgbuf=1048576` we need another command, please add it to your instructions: `sudo dmesg | grep -i intelfir`

Here is the log, anything you can read with your experience?

With **IntelBluetoothInjector.kext** enabled:
```
[    1.199313]: IntelFirmware: Driver init()
[    1.199318]: IntelFirmware: Driver Probe()
[    1.199323]: IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0025
[    1.199360]: IntelFirmware: Driver Start()
[    2.660088]: Sandbox: 1 duplicate report for dirhelper deny(1) file-write-unlink /private/var/folders/zz/zyxvpxvq6csfxvn_n00000sm00006d/T/.CalendarLocksSandbox: contextstored(104) deny(1) user-preference-read kCFPreferencesAnyApplicationsuccessfully truncated syncroots to 0 bytessuccessfully truncated syncroot_ancestors to 0 bytessuccessfully truncated network-exemptions to 0 bytesSandbox: 1 duplicate report for contextstored deny(1) user-preference-read kCFPreferencesAnyApplicationSandbox: distnoted(120) deny(1) file-read-metadata /usr/sbin/distnotedSandbox: distnoted(120) deny(1) file-read-metadata /usrIntelFirmware: set configuration to 1
[    2.700492]: IntelFirmware: Found interface!!!
[    2.700514]: IntelFirmware: Found Interrupt endpoint!
[    2.700706]: IntelFirmware: Found Bulk out endpoint!
[    2.700872]: IntelFirmware: Found Bulk in endpoint!
[    2.701068]: IntelFirmware: usb init succeed
[    2.701072]: IntelFirmware: HCI_OP_INTEL_VERSION
[    2.701287]: IntelFirmware: interrupt wait
[    2.701568]: IntelFirmware: Firmware revision 0.0 build 111 week 53 2019IntelFirmware: suspect device firmware: ibt-19-16-0.sfi
[    2.701576]: IntelFirmware: firmware had been download.
[    2.701584]: IntelFirmware: interrupt continue
[    2.701587]: IntelFirmware: HCI_OP_INTEL_EVENT_MASK
[    2.701761]: IntelFirmware: interrupt wait
[    2.702559]: IntelFirmware: interrupt continue
[    2.702567]: IntelFirmware: End download
[    2.702570]: IntelFirmware: Clean up...
```

With **IntelBluetoothInjector.kext** disabled:
```
[    1.123112]: IntelFirmware: Driver init()
[    1.123118]: IntelFirmware: Driver Probe()
[    1.123121]: IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0025
[    1.123159]: IntelFirmware: Driver Start()
[    2.623754]: IntelFirmware: set configuration to 1
[    2.624112]: IntelFirmware: Found interface!!!
[    2.624131]: IntelFirmware: Found Interrupt endpoint!
[    2.624296]: IntelFirmware: Found Bulk out endpoint!
[    2.624483]: IntelFirmware: Found Bulk in endpoint!
[    2.624648]: IntelFirmware: usb init succeed
[    2.624652]: IntelFirmware: HCI_OP_INTEL_VERSION
[    2.624823]: IntelFirmware: interrupt wait
[    2.625080]: IntelFirmware: Firmware revision 0.0 build 111 week 53 2019IntelFirmware: suspect device firmware: ibt-19-16-0.sfi
[    2.625086]: IntelFirmware: firmware had been download.
[    2.625092]: IntelFirmware: interrupt continue
[    2.625095]: IntelFirmware: HCI_OP_INTEL_EVENT_MASK
[    2.625250]: IntelFirmware: interrupt wait
[    2.626081]: IntelFirmware: interrupt continue
[    2.626087]: IntelFirmware: End download
[    2.626089]: IntelFirmware: Clean up...
```

**P.S.** I think you miss a line-feed character after "Firmware revision" etc. as you can see the line is glued with the next one. Also the "Firmware revision" is mentioned **0.0** is this normal?

Thank you...