OpenIntelWireless/IntelBluetoothFirmware #311: Bluetooth Firmware loading

gnpaone, 2021-07-10:
Until Big Sur, I used to use IntelBluetoothFirmware.kext and IntelBluetoothInjector.kext (attached below). Now I have upgraded to Monterey but the bluetooth stopped working (firmware doesn't load, bluetooth turned off, bluetooth toggle switch not working), I searched the internet for a solution when I came across your page, it says replace IntelBluetoothInjector.kext with BlueToolFixup.kext (as mentioned in this "[Instructions](https://openintelwireless.github.io/IntelBluetoothFirmware/FAQ.html#what-additional-steps-should-i-do-to-make-bluetooth-work-on-macos-monterey)" page), then the bluetooth did turn on and off and it got detected in system info > bluetooth, but it is not discovering any bluetooth devices and hackintool shows firmware not loaded but it shows loaded in big sur. Can you please compile BlueToolFixup.kext to support intel too? I use intel AX200 wifi+bluetooth device (wifi is working, thanks to your beta version AirportItlwm.kext - for Monterey, as it is macos beta version I updated all my kexts to beta for it to work as some of  the release versions didn't work so I also tried with your latest beta bluetooth kext as well as latest release version (1.1.3) of bluetooth kext). 
[IntelBluetooth_gnpaone.zip](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/6794932/IntelBluetooth_gnpaone.zip)

closed 2021-07-12 by zxystd

stevezhengshiqi, 2021-07-12:
BlueToolFixup.kext supports Intel BT without any problem. There's no such thing that "compile BlueToolFixup.kext to support intel". Make sure you use the latest Lilu + BlueToolFixup.kext with all the other latest kexts.

gnpaone, 2021-07-12:
@stevezhengshiqi are you sure? Will it work for intel ax200?

stevezhengshiqi, 2021-07-12:
> @stevezhengshiqi are you sure? Will it work for intel ax200?

No doubt.

gnpaone, 2021-07-12:
@stevezhengshiqi Ok, I figured out what's wrong. My bluetooth is working but a slight problem (I think it's a bug maybe). Here is what I found:
I booted into my macos and opened settings to scan for my bluetooth headset, I saw that my headset appears in the list for a fraction of second (like just a blink of eye time) and vanishes, again it appears for fraction of second after few seconds interval and disappears. Then I restarted my mac and the problem is attached in the pics below (The pics are after reboot) it does show my headset this time which is in one of the pic but it's greyed out, even this time it does ungreys for a fraction of second, now I am so accurate to click on my headset option as shown in other screenshot, it does load to connect but doesn't connect and greys out again, also the bluetooth settings doesn't load after reboot.

Common thing: Hackintool shows firmware not loaded but it shows loaded in Big sur. And also my bluetooth chip in my laptop is connected well and working fine in big sur and windows.

[image]
[image]
[image]
[image]

MrSu0215, 2021-07-19:
This article has a solution
https://github.com/OpenIntelWireless/IntelBluetoothFirmware/issues/316

gnpaone, 2021-07-19:
> This article has a solution
> #316

@MrSu0215 Thank you 😊