OpenIntelWireless/IntelBluetoothFirmware #312: Kernel panic due to intelbluetoothfirmware

xunkutech, 2021-07-14:
**Description**
After installing the latest version, the system hangs when it tries to sleep.

**Environment**

- Kext Version: pre-2.0.0
- WiFi Card Model: 7260
- Product ID: 
- macOS Version: 11.4

**Info in System Report - Bluetooth**

```
  Apple Bluetooth Software Version:	8.0.5d7
  Hardware, Features, and Settings:
  Name: xxx's iMac
  Address:	4C-EB-42-xx-xx-xx
  Bluetooth Low Energy Supported:	Yes
  Handoff Supported:	Yes
  Instant Hot Spot Supported:	Yes
  Manufacturer:	Intel
  Transport:	USB
  Firmware Version:	6.3584 (6.3584)
  Bluetooth Power:	On
  Discoverable:	Off
  Connectable:	Yes
  Auto Seek Pointing:	On
  Remote wake:	On
  Vendor ID:	0x8087
  Product ID:	0x07DC
  Bluetooth Core Spec:	4.0 (0x6)
  HCI Revision:	0xE00
  LMP Version:	4.0 (0x6)
  LMP Subversion:	0xE00
  Device Type (Major):	Computer
  Device Type (Complete):	Mac Desktop
  Composite Class Of Device:	0x380104
  Device Class (Major):	0x01
  Device Class (Minor):	0x01
  Service Class:	0x1C0
  Auto Seek Keyboard:	On
  Devices (Paired, Configured, etc.):
  Keyboard K380:
  Address:	34-88-5D-42-C8-C4
  Major Type:	Peripheral
  Minor Type:	Keyboard
  Services:	Multi-Device Keyboard K380     , Multi-Device Keyboard K380     , Multi-Device Keyboard K380     
  Paired:	Yes
  Configured:	Yes
  Connected:	No
  Manufacturer:	Broadcom (0x5, 0x2304)
  Bluetooth Core Spec:	3.0
  Firmware Version:	0x4200
  Vendor ID:	0x046D
  Product ID:	0xB342
  Class of Device:	0x05 0x10 0x0540
  Host Connectable:	No
  EDR Supported:	No
  eSCO Supported:	No
  SSP Supported:	Yes
  Services:	Hands-Free unit, Headset
  Paired:	Yes
  Configured:	Yes
  Connected:	No
  Manufacturer:	Cambridge Silicon Radio (0x7, 0x21C8)
  Bluetooth Core Spec:	4.1
  Class of Device:	0x04 0x01 0x240404
  EDR Supported:	Yes
  eSCO Supported:	Yes
  SSP Supported:	Yes
  Services:
  Bluetooth File Transfer:
  Folder other devices can browse:	~/Public
  When receiving items:	Accept all without warning
  State:	Disabled
  Bluetooth File Exchange:
  Folder for accepted items:	~/Downloads
  When other items are accepted:	Save to location
  When receiving items:	Accept all without warning
  State:	Disabled
  Bluetooth Internet Sharing:
  State:	Disabled
  Incoming Serial Ports:
  Bluetooth-Incoming-Port:
  RFCOMM Channel:	3
  Requires Authentication:	No

```

**Logs**

```log
2021-07-14 16:00:05.918505+0800 0x2ee      Default     0x0                  86     0    kernelmanagerd: Received kext load notification: com.zxystd.IntelBluetoothFirmware
```

**Kernel Panic Logs**

```
panic(cpu 0 caller 0xffffff801c44f642): "Kernel stack memory corruption detected"@/System/Volumes/Data/SWE/macOS/BuildRoots/e90674e518/Library/Caches/com.apple.xbs/Sources/xnu/xnu-7195.121.3/libkern/stack_protector.c:37
Backtrace (CPU 0), Frame : Return Address
0xffffffa0b25db5b0 : 0xffffff801c48e0dd 
0xffffffa0b25db600 : 0xffffff801c5d4f33 
0xffffffa0b25db640 : 0xffffff801c5c552a 
0xffffffa0b25db690 : 0xffffff801c432a2f 
0xffffffa0b25db6b0 : 0xffffff801c48d8fd 
0xffffffa0b25db7d0 : 0xffffff801c48dbf3 
0xffffffa0b25db840 : 0xffffff801cc9d81a 
0xffffffa0b25db8b0 : 0xffffff801c44f642 
0xffffffa0b25db8c0 : 0xffffff801cb88c26 
0xffffffa0b25dbb70 : 0xffffff801cb8903d 
0xffffffa0b25dbb90 : 0xffffff801cbafd7e 
0xffffffa0b25dbd40 : 0xffffff80204e2c88 
0xffffffa0b25dbd60 : 0xffffff80204e2bf7 
0xffffffa0b25dbdd0 : 0xffffff801cbc591b 
0xffffffa0b25dbe40 : 0xffffff801cbc54ae 
0xffffffa0b25dbef0 : 0xffffff801cbc464f 
0xffffffa0b25dbf50 : 0xffffff801cbc73c6 
0xffffffa0b25dbfa0 : 0xffffff801c43213e 
      Kernel Extensions in backtrace:
         com.zxystd.IntelBluetoothFirmware(2.0)[929D6B5E-37D7-31F2-9EE5-60D3DE652541]@0xffffff80204de000->0xffffff8020793fff
            dependency: com.apple.iokit.IOUSBHostFamily(1.2)[BD7179E5-F2E2-3E86-83D5-F3EA5EE1EE3F]@0xffffff801f288000->0xffffff801f316fff

Process name corresponding to current thread: kernel_task
Boot args: chunklist-security-epoch=0 -chunklist-no-rev2-dev

Mac OS version:
20F71

Kernel version:
Darwin Kernel Version 20.5.0: Sat May  8 05:10:33 PDT 2021; root:xnu-7195.121.3~9/RELEASE_X86_64
Kernel UUID: 52A1E876-863E-38E3-AC80-09BBAB13B752
KernelCache slide: 0x000000001c200000
KernelCache base:  0xffffff801c400000
Kernel slide:      0x000000001c210000
Kernel text base:  0xffffff801c410000
__HIB  text base: 0xffffff801c300000
System model name: iMac15,1 (Mac-42FD25EABCABB274)
System shutdown begun: NO
Panic diags file available: YES (0x0)
Hibernation exit count: 0

System uptime in nanoseconds: 82196105779351
Last Sleep:           absolute           base_tsc          base_nano
  Uptime  : 0x00004ac1c5afdbfe
  Sleep   : 0x00004abee599ae6d 0x0000000032288b66 0x000049e6fbbe1f15
  Wake    : 0x00004abee61461a5 0x000000003140eae0 0x00004abee5ae66d5
```

williambj1, 2021-07-15:
Does it happen every time? Also, please reattach the panic logs with `keepsyms=1` in boot-args.

xunkutech, 2021-07-15:
> Does it happen every time? Also, please reattach the panic logs with `keepsyms=1` in boot-args.

It happens when I leave the machine for a long time and come back to use it again. It seems that something prevents the OS from going to sleep. I've reverted the driver to version 1.1.2 which seems more stable for me(also fixed the sleeping issue).

williambj1, 2021-07-15:
The main goal here is to figure out the cause of the panic, the driver got a complete rewrite so the previous versions don't matter anymore.

Could you add `keepsyms=1` in boot-args and attach the panic logs when it triggers? Much appreciated.

BTW, the RC version is not built from the latest commit, have you tried the latest? Available on the GitHub Actions Artifacts page.

xunkutech, 2021-07-16:
> The main goal here is to figure out the cause of the panic, the driver got a complete rewrite so the previous versions don't matter anymore.
>
> Could you add `keepsyms=1` in boot-args and attach the [...]

I downloaded the driver from CI. Should I use the DEBUG version to dump the log?

williambj1, 2021-07-16:
Yes, the debug version is required.

singhalrishi27, 2021-08-12:
Hello bro I have the same card as you have when I use intel Bluetooth firmware 1.3 everything just works fine but when I update kext to 2.0 it doesn't work in system preferences Bluetooth is turned off by default and I can't turn it back on same behaviour applies to macos Big Sur I wasn't able to obtain logs it always came out empty can you please check if the kext loads or not

singhalrishi27, 2021-08-12:
#303