OpenIntelWireless/IntelBluetoothFirmware #315: Bluetooth completely stopped working after installing last beta version of driver

SNPR, 2021-07-18:
**Description**
I have Asus TUF Gaming Z490 Plus Wi-Fi motherboard, this motherboard has integrated Bluetooth and I use it. After installing v2.0.0 RC1 kext I made a reboot. After reboot Bluetooth worked properly. Then I send the computer to sleep. After waking up Bluetooth completely stopped working (crossed Bluetooth icon).
I also have Windows 10 on another drive, so I tried to boot to Windows 10 and in Windows Bluetooth didn't work too (Bluetooth device disappeared in device manager).
Maybe it's just a coincidence but I suppose this driver somehow killed my Bluetooth.

I tried to reset NVRAM, install the last stable version of IntelBluetoothFirmware kext and made reset the Bluetooth module - nothing help.

**Environment**

- Kext Version: v2.0.0 RC1
- WiFi Card Model: Intel® WiFi 6 – AX201
- Product ID: 
- macOS Version: 11.4

**Info in System Report - Bluetooth**

```
![image](https://user-images.githubusercontent.com/22043414/126072785-08665788-adec-433d-8279-de8e68e9cb40.png)

```

**Logs**

```log
empty log
```

**Screenshots**
[image]
[image]
[image]

williambj1, 2021-07-18:
Have you tried Linux?

SNPR, 2021-07-18:
> Linux

No, I’m not tried to install Linux. 

I also tried to update my motherboard’s bios - didn’t help.

MrSu0215, 2021-07-19:
#316 You can refer to this article to solve it.

zxystd, 2021-07-19:
@SNPR 
delete all of the Intel BT drivers and see if it is the same.

SNPR, 2021-07-20:
@MrSu0215, @zxystd, unfortunately, nothing of this helped me :-(

MrSu0215, 2021-07-20:
Set up your usb mapping

SNPR, 2021-07-20:
@MrSu0215 I already set up my USB mapping before this issue.
I just tried this again and it worked! https://github.com/OpenIntelWireless/IntelBluetoothFirmware/issues/316
The first time, when I tried to follow these steps I used the last beta version of the Bluetooth driver.
Now I tried to use the last stable version and it helped.

Thanks, everyone!

closed 2021-07-20 by SNPR