OpenIntelWireless/IntelBluetoothFirmware #316: Solution to Bluetooth loss when using Intel AX200 network card

MrSu0215, 2021-07-18:
Solution to Bluetooth loss when using Intel AX200 network card
因特尔AX200网卡使用时蓝牙丢失的解决办法
Install and use AX200 network card, Wi-Fi is normal but the Bluetooth control panel disappears or the Bluetooth switch cannot be used.
安装使用AX200网卡，Wi-Fi正常但蓝牙控制面板消失不见或者蓝牙开关无法使用。
To make Intel Bluetooth work in Big Sur, the following are required:

1. The latest version of Lilu (from source code)
2. The latest version of BrcmPatchRAM (from source code), this package only needs
BlueToolFixup.kext
3. Download and install the following files in the author's Bluetooth driver package
IntelBluetoothFirmware.kext
IntelBluetoolInjector.kext.
4. Turn off XhciPortLimit in OpenCore Configurator
5. Save the config file
6. Shut down and restart the machine instead of restarting it.
要使英特尔蓝牙在Big Sur工作，需要以下内容：

1.最新版的Lilu（来自源码）
2.最新版本的 BrcmPatchRAM（来自源码），这个包中只需要
BlueToolFixup.kext
3.下载安装作者蓝牙驱动包中的一下文件
IntelBluetoothFirmware.kext
IntelBluetoolInjector.kext。
4.在OpenCore Configurator中关闭 XhciPortLimit
5.保存config文件
6.关闭并重新启动机器而不是重新启动。
[image]
After booting, you can see the Bluetooth hardware information in the hackintool and system reports. During use, Bluetooth is normal and the system's Bluetooth switch is operating normally.
开机后，便可以在hackintool和系统报告中看到蓝牙的硬件信息。在使用过程中，蓝牙正常且系统蓝牙开关正常操作。
[image]
[image]

gnpaone, 2021-07-19:
@MrSu0215 So should I keep intelbluetoothinjector.kext too alongwith intelbluetoothfirmware.kext and bluetoolfixup.kext or should I remove it?

MrSu0215, 2021-07-19:
> @ MrSu0215那么我应该将 intelbluetoothinjector.kext 与 intelbluetoothfirmware.kext 和 bluetoolfixup.kext 一起保留还是应该删除它？

MrSu0215, 2021-07-19:
> @ MrSu0215 那我应该将 intelbluetoothinjector.kext 与 intelbluetoothfirmware.kext 和 bluetoolfixup.kext 一起保留还是应该删除它？

Keep together
一起保留

gnpaone, 2021-07-19:
@MrSu0215 Thanks for the solution ☺️. It's working well. Finally I found the solution 🤩.

MrSu0215, 2021-07-19:
> @MrSu0215 Thanks for the solution ☺️. It's working well. Finally I found the solution 🤩.

If you have any questions during use, you can contact me by private message

NuitSombre77, 2021-07-29:
It's not working for me, someone can help me pls ?

gnpaone, 2021-07-31:
> It's not working for me, someone can help me pls ?

Hey @NuitSombre77  Actually this workaround works partially for intel ax200 bluetooth. I will explain it in next few sentences. I got my bluetooth working with "intelbluetoothfirmware.kext + bluetoolfixup.kext alongwith XhciPortLimit False (thanks to @MrSu0215 ), also *remove* intelbluetoothinjector.kext (bluetooth doesn't work with this kext for macos monterey)". As I mentioned earlier, it works partially, i.e. if you have already paired any bluetooth devices before upgrading to monterey (in Big Sur) then you can flawlessly connect those devices without any problem, but new bluetooth devices are not getting detected (bluetooth discovery mode is turned off - checked via about this mac > system report > bluetooth), this is the problem. Can anyone please help me solve this problem

wangzikang-debug, 2021-08-06:
[image]
跟这个没关系，你USB定制没，我的AX200完全没问题

wangzikang-debug, 2021-08-06:
> [image]
> これは关系ありません、あなたのUSBはカスタマイズされています、私のAX200は完全に问题ありません。

MrSu0215, 2021-08-06:
> > [image]
> > これは关系ありません、あなたのUSBはカスタマイズされています、私のAX200は完全に问题ありません。

私が与えた解決策は参照解決策です。あなたの解決策が正常であれば、それを参照する必要はありません。

gnpaone, 2021-08-06:
> > It's not working for me, someone can help me pls ?
> 
> Hey @NuitSombre77 First it didn't work for me #311 (I uploaded my initial screenshots and recording in that) But this workaround actually works partially for intel ax200 bluetooth. I will explain it in next few sentences. I got my bluetooth working with "intelbluetoothfirmware.kext + bluetoolfixup.kext alongwith XhciPortLimit False (thanks to @MrSu0215 ), also _remove_ intelbluetoothinjector.kext (bluetooth doesn't work with this kext for macos monterey)". As I mentioned earlier, it works partially, i.e. if you have already paired any bluetooth devices before upgrading to monterey (in Big Sur) then you can flawlessly connect those devices without any problem, but new bluetooth devices are not getting detected (bluetooth discovery mode is turned off and not turning on - checked via about this mac > system report > bluetooth), this is the problem. Can anyone please help me solve this problem

Update: Bluetooth works and detects other devices most of the times but sometimes it fails to detect new devices. I consider it as a success tho

gnpaone, 2021-08-10:
> AirportItlwm.kext + BluetoothFixup.kext + IntelBluetoothFirmware.kext working fine on Monterey

Yah yah, it's working fine

ftitreefly, 2021-08-15:
There's two USB built-in socket, USB1 and USB2, on my mainbord. Some day I switched the two sockets, problem solved.

Hanly-Wijaya, 2021-08-23:
Do I add IntelBluetoothInjector or do I do, Bluetoolfixup & IntelBluetoothFirmware? I've been trying the 2nd method (bluetoolfixup & intelbluetoothfirmware) but bluetooth is still not functioning. I didn't keep note of what model of bluetooth I have btw so I'll reply back when I find it.

Hanly-Wijaya, 2021-08-23:
Update: I do not know what bluetooth card I have but I do know that i have an AX1650i network card if that helps. I tried looking for model of bluetooth through Ubuntu but it didn't show anything. Will install Windows and see what model of bt it is.

gnpaone, 2021-08-23:
> Do I add IntelBluetoothInjector or do I do, Bluetoolfixup & IntelBluetoothFirmware? I've been trying the 2nd method (bluetoolfixup & intelbluetoothfirmware) but bluetooth is still not functioning. I didn't keep note of what model of bluetooth I have btw so I'll reply back when I find it.

You can check bluetooth card model using hackintool if you are using hackintosh

hanlywijaya, 2021-08-23:
I checked in Hackintool and it doesn't show up.

zxystd, 2021-08-23:
do USB mapping.

hanlywijaya, 2021-08-23:
Just wasted my time installing windows lol. Just saying intel wireless bluetooth

hanlywijaya, 2021-08-23:
> do USB mapping.

Sorry, didn't see your comment. I already did usb mapping properly, didn't show anything about bluetooth so I tried it and same thing happens.

gnpaone, 2021-08-23:
> I checked in Hackintool and it doesn't show up.

No, it will show up for sure, click other peripherals and send screenshot. Make sure intelbluetoothfirmware and bluetool is present in your EFI

hanlywijaya, 2021-08-23:
> > I checked in Hackintool and it doesn't show up.
> 
> No, it will show up for sure, click other peripherals and send screenshot. Make sure intelbluetoothfirmware and bluetool is present in your EFI

It's nearly 12am here, I'll post when I get up next morning

williambj1, 2021-08-23:
Use other sources for personal help.