OpenIntelWireless/IntelBluetoothFirmware #321: Intel Wifi and Bluetooth don't work together on AC 3165

FammasMaz, 2021-08-05:
**Description**

**Environment**

- Kext Version: 1.1.3
- WiFi Card Model: AC 3165
- Product ID: 0x0A2A
- macOS Version: 11.5

**Info in System Report - Bluetooth**

```
Apple Bluetooth Software Version:	8.0.5d7
  Hardware, Features and Settings:
  Name:	MacBook Pro
  Address:	DC-53-60-5F-76-3D
  Bluetooth Low Energy Supported:	Yes
  Handoff Supported:	Yes
  Instant Hot Spot Supported:	Yes
  Manufacturer:	Intel
  Transport:	USB
  Firmware Version:	8.4096 (8.4096)
  Bluetooth Power:	On
  Discoverable:	Off
  Connectable:	Yes
  Auto Seek Pointing:	On
  Remote wake:	On
  Vendor ID:	0x8087
  Product ID:	0x0A2A
  Bluetooth Core Spec:	4.2 (0x8)
  HCI Revision:	0x1000
  LMP Version:	4.2 (0x8)
  LMP Subversion:	0x1000
  Device Type (Major):	Computer
  Device Type (Complete):	Mac Portable
  Composite Class Of Device:	0x38010C
  Device Class (Major):	0x01
  Device Class (Minor):	0x03
  Service Class:	0x1C0
  Auto Seek Keyboard:	On
  Devices (Paired, Configured, etc.):
  AirPods pro:
  Address:	E5-00-1D-1E-38-80
  Major Type:	Audio
  Minor Type:	Headset
  Services:	JL_A2DP, AAP Server, JL_HFP
  Paired:	Yes
  Configured:	Yes
  Connected:	No
  Bluetooth Core Spec:	5.1
  Firmware Version:	0​B​896
  Vendor ID:	0x004C
  Product ID:	0x2002
  Class of Device:	0x04 0x01 0x240404
  EDR Supported:	Yes
  eSCO Supported:	Yes
  SSP Supported:	Yes
  Redmi Note 7:
  Address:	22-22-01-C1-83-7A
  Major Type:	Phone
  Minor Type:	Smartphone
  Services:	
  Paired:	Yes
  Configured:	Yes
  Connected:	No
  Manufacturer:	Unknown (0x9, 0x2BE)
  Bluetooth Core Spec:	5.0
  Class of Device:	0x02 0x03 0x7A020C
  EDR Supported:	Yes
  eSCO Supported:	Yes
  SSP Supported:	Yes
  Services:
  Bluetooth File Transfer:
  Folder other devices can browse:	~/Public
  When receiving items:	Accept all without warning
  State:	Disabled
  Bluetooth File Exchange:
  Folder for accepted items:	~/Downloads
  When other items are accepted:	Save to location
  When receiving items:	Accept all without warning
  State:	Disabled
  Bluetooth Internet Sharing:
  State:	Disabled
  Incoming Serial Ports:
  Bluetooth-Incoming-Port:
  RFCOMM Channel:	3
  Requires Authentication:	No

```

**Logs**

```log

```

**Kernel Panic Logs**

```

```

**Screenshots**

zxystd, 2021-08-05:
I have two, and they work fine.

closed 2021-08-05 by zxystd

FammasMaz, 2021-08-05:
> I have two, and they work fine.

Could this be because I use airportitlmw and not itlmw with heliport? Also can you please tell the version of both kexts you are using in which they work at the same time? My bluetooth is mapped under usb3.0. can that cause an issue?

williambj1, 2021-08-05:
Try swapping the antenna and see if it helps.

FammasMaz, 2021-08-05:
> Try swapping the antenna and see if it helps.

Sorry, it didn't. I have the same issue as people on this thread:

https://github.com/OpenIntelWireless/itlwm/issues/482#issuecomment-892172847