OpenIntelWireless/IntelBluetoothFirmware #322: a setting in config.plist with later than 2.0 will cause stuck and showing EB at booting

ske1996, 2021-08-08:
**Description**
I've got a stuck at booting after replacing injector and firmware kexts to 2.0 stable from 1.3
when i enabled -v and it showed me:
[image]
and this is my config,the part about kexts:
[image]

I fixed that EB issue with removing blue circle part of upper image.(set it ad blank)
but that setting worked just fine with version 1.3.

Just sharing a experience. 
Not a bug reporting.

**Environment**

- Kext Version: 2.0 stable
- WiFi Card Model: 9560ac
- Product ID: 8086
- macOS Version: 11.4
- OpenCore Version:0.6.5

MrSu0215, 2021-08-08:
Check if the files belonging to your driver are placed in the kexts folder under efi.

ske1996, 2021-08-08:
> Check if the files belonging to your driver are placed in the kexts folder under efi.

……i fixed that,
just share what did i do…
not for requesting help😂😂😂

williambj1, 2021-08-20:
Thanks, I've pinned this guide and adjusted the title to make It fit well.

closed 2021-08-20 by williambj1