OpenIntelWireless/IntelBluetoothFirmware #325: When macOS Monterey boot, it still loads in a loop, causing slow startup

Luastonsu, 2021-08-12:
**Description**

I installed macOS Monterey Beta 5 and used the latest IntelBluetoothFirmware.kext and IntelBluetoothInjector.kext. When I start Monterey after disabling these two kexts in config.plist, the startup is very fast. But when I start Monterey after enabling these two kexts, the boot speed will be very slow, and it will keep reporting "AppleKeyStore:11150:150:operation failed.(sel:7 ret:e00002c2,-1,100000)". Usually the boot stuck here cannot continue for several minutes.
**Environment**

- Kext Version: v2.0.0 Stable
- WiFi Card Model: AC3165
- Product ID: unknown
- macOS Version: macOS Monterey Beta 5

**Info in System Report - Bluetooth**

```

```

**Logs**

```log

```

**Kernel Panic Logs**

```

```

**Screenshots**

Luastonsu, 2021-08-12:
Please fix this problem as soon as possible, it has already affected the use.
Thanks in advance.

zxystd, 2021-08-12:
why not read our docs or the release notes?

closed 2021-08-12 by zxystd

Luastonsu, 2021-08-13:
> why not read our docs or the release notes?

I am sorry.I started use GitHub before some months.There are too much things I am understand.

williambj1, 2021-08-13:
Doesn't seemed to directly related to the firmware uploader. It may take longer to enter desktop since the bluetooth stack from Apple needs time to initialize.

Luastonsu, 2021-08-13:
> why not read our docs or the release notes?

But after I used BlueTooFixup.kext,Bluetooth will doesn't open or off on macOS Big Sur.

williambj1, 2021-08-13:
BlueTool does not even exist in Big Sur. I would suggest reading the docs carefully one more time.