OpenIntelWireless/IntelBluetoothFirmware #329: mac os beta 3 intel n7265 bluetooth4.2 iuse IntelBluetoothFirmware.kext. BlueToolFixup.kext not work good

edwardwich, 2021-08-16:
**Description**
mac os beta 3 intel n7265 bluetooth4.2 iuse IntelBluetoothFirmware.kext. BlueToolFixup.kext not work good
**Environment**

- Kext Version: v2.0.0 stable and 1.1.3 & BlueToolFixup.kext
- WiFi Card Model: Intel Corporation, Wireless 7265
- Product ID: Vendor ID:	0x8086.   ,Device ID:	0x095a  

- macOS Version: macOS Monterey Beta3

**Info in System Report - Bluetooth**
I can't tern off Bluetooth always on bluetooth and not stable
**Screenshots**
[image]
[image]

williambj1, 2021-08-16:
Sorry but nothing could be advised based on the given information. Did it work well before?

closed 2021-08-16 by williambj1

Edwardwich, 2021-08-31:
> Sorry but nothing could be advised based on the given information. Did it work well before?

work good on Big Sur