OpenIntelWireless/IntelBluetoothFirmware #332: AC8265 Bluetooth not working

1314mjf521, 2021-08-21:
小米Air13.3蓝牙无法驱动
 系统版本：12.0 Beta版(21A5304g)

[image]
[image]
[image]
[image]

LOG：
`[  141.454173]: **** [IOBluetoothHCIUserClient][SetBTLPOffWL] -- mBluetoothFamily->mACPIMethods->SetBTLP (OFF) returned 0xE00002C7 error
[  141.454179]: **** [IOBluetoothHCIUserClient][SetBTLPOffWL] -- mBluetoothFamily->mACPIMethods->SetBTLP (OFF) returned 0xE00002C7 error
[  141.454243]: **** [IOBluetoothHCIUserClient][SetBTRSWL] -- mBluetoothFamily->mACPIMethods->SetBTRS() returned 0xE00002C7 error
[  141.454247]: **** [IOBluetoothHCIUserClient][SetBTRSWL] -- mBluetoothFamily->mACPIMethods->SetBTRS() returned 0xE00002C7 error
[  141.485196]: Sandbox: mobileassetd(310) deny(1) file-read-metadata /Library**** [IOBluetoothHCIUserClient][SetBTLPOffWL] -- mBluetoothFamily->mACPIMethods->SetBTLP (OFF) returned 0xE00002C7 error
[  141.488994]: **** [IOBluetoothHCIUserClient][SetBTLPOffWL] -- mBluetoothFamily->mACPIMethods->SetBTLP (OFF) returned 0xE00002C7 error
[  141.489050]: **** [IOBluetoothHCIUserClient][SetBTPUWL] -- mBluetoothFamily->mACPIMethods->SetBTPU() returned 0xE00002C7 error
[  141.489053]: **** [IOBluetoothHCIUserClient][SetBTPUWL] -- mBluetoothFamily->mACPIMethods->SetBTPU() returned 0xE00002C7 error`

1314mjf521, 2021-08-21:
[image]
这是加载的驱动

williambj1, 2021-08-21:
请提供完整日志，问题并没有出在这里

1314mjf521, 2021-08-21:
[image]
这个里面的介绍的驱动无法安装，安装完以后直接黑屏，有没有其他办法提供日志？

1314mjf521, 2021-08-21:
[NewFile-1.txt](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/7025619/NewFile-1.txt)
[NewFile.txt](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/7025620/NewFile.txt)
这是两个完整的系统日志不知道可不可以？

singhalrishi27, 2021-08-24:
same issue

singhalrishi27, 2021-08-29:
In the Latest Commit issue was fixed but it has worse audio quality and headphones are unusable when mic input is set to bluetooth can you check if you are having the same issue

gujiangjiang, 2021-09-21:
我能驱动，但是在Monterey下蓝牙关了就开不了，只能重启。。。

totor-iot, 2021-09-25:
In Monterey(21A5522h), bluetooth injector can not injecting the device(8265) and it's causing to occur long boot up time. Also bluetooth doesn't work.
[image]

gujiangjiang, 2021-09-25:
@totor-iot 
What additional steps should I do to make Bluetooth work on macOS Monterey

Make sure IntelBluetoothFirmware.kext v1.1.3 or newer is used.
Exclude the injection of IntelBluetoothInjector.kext for macOS Monterey from the Bootloader's config
Include BlueToolFixup.kext from acidanthera/BrcmPatchRAM

williambj1, 2021-09-25:
Apple is creating even more nasty bugs starting from macOS Monterey B6, make sure you are using a SMBIOS model before 2019.

closed 2021-09-25 by williambj1

totor-iot, 2021-09-25:
> @totor-iot
> What additional steps should I do to make Bluetooth work on macOS Monterey
>
> Make sure IntelBluetoothFirmware.kext v1.1.3 or newer is used.
> Exclude the injection of [...]

I applied this method except one condition. I used BlueToolFixup.kext but didn't exclude Injector, so my application was like this:
1- BlueToolFixup.kext
2- IntelBluetoothFirmware.kext
3- IntelBluetoothInjector.kext

But as you suggest, it should be like this:
1- BlueToolFixup.kext
2- IntelBluetoothFirmware.kext

Anyway i turned back to Big Sur
Thank you for your help.

gujiangjiang, 2021-09-25:
@totor-iot 
Monterey dont need Injector kexts, just delete it.

1314mjf521, 2021-09-26:
Monterey  B7 蓝牙依旧不工作，加载方式跟上面一样，smbios设置为17.1不工作，我在尝试换其他的模型看看能不能工作吧

gujiangjiang, 2021-09-26:
@1314mjf521 
我是B7，正常工作，但是无法关闭，关闭了就不能再开了，只能重启

williambj1, 2021-09-26:
> @1314mjf521
> 我是B7，正常工作，但是无法关闭，关闭了就不能再开了，只能重启

```sh
sudo killall bluetoothd
```

Please ask Apple to fix this trash.