OpenIntelWireless/IntelBluetoothFirmware #338: bluetooth work but when tern off I can't tern on need rest N-7265

Edwardwich, 2021-09-02:
**Description**
I'm update Mac OS to Mac OS 12.0 Beta (21A5304g) bluetooth work very bad 

**Environment**

- Kext Version: v2.0.0 stable and BlueToolFixup.kext
- WiFi Card Model: n7265 m.2
- macOS Version:  12.0 Beta (21A5304g)

**Info in System Report - Bluetooth**
bluetooth work but when tern off I can't tern on need rest my laptop 

```

```

**Logs**

```

```

**Kernel Panic Logs**

```

```

**Screenshots**

[image]
[image]

after tern off bluetooth can't tern on 
[image]

baraach, 2021-09-02:
Does it work fast or slow for you? I am using my mouse with dongle instead of bluetooth because it works so slow with bluetooth. I have AC 3165.

williambj1, 2021-09-02:
https://github.com/acidanthera/bugtracker/issues/1771#issuecomment-912053064

closed 2021-09-02 by williambj1