OpenIntelWireless/IntelBluetoothFirmware #346: After update to Mac OS 12beta 10 Bluetooth tern off and never on

Edwardwich, 2021-10-13:
**Description**
<After update to Mac OS 12beta 10 Bluetooth tern off and never on >

**Environment**

- Kext Version: I test 2.01 ,2.0,1.3
- WiFi Card Model: N7265
- Product ID: 0x8087 ,0x0A2A
- macOS Version: 12.0 Beta (21A5552a)

**Info in System Report - Bluetooth**
<After update to Mac OS 12beta 10 Bluetooth tern off and never back to on  >

```

```

**Logs**

```log

```

**Kernel Panic Logs**

```

```

**Screenshots**

[image]

[image]
[image]
[image]
[IOReg.zip](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/7341541/IOReg.zip)

G0LD, 2021-10-13:
Yep. Can confirm Beta 10 broke bluetooth compatibility
[image]

williambj1, 2021-10-13:
Duplicate of #338

closed 2021-10-13 by williambj1