OpenIntelWireless/IntelBluetoothFirmware #353: Monterey doesn't work

shangjun2x, 2021-10-30:
**Description**
Monterey does't work on the intel bluetooth, Big Sur works well.

**Environment**
Thinkpad yoga X1C7

- Kext Version: 2.0.1
- WiFi Card Model: WiFi works well
- Product ID: 
- macOS Version: Monterey 12.0.1

**Info in System Report - Bluetooth**

[image]

[image]
```

```

**Logs**

```log

```

**Kernel Panic Logs**

```

```

**Screenshots**

williambj1, 2021-10-30:
Use the latest version of BlueToolFixup compiled from source

closed 2021-10-30 by williambj1

VictorLeonardoFranklin, 2021-10-30:
> Use the latest version of BlueToolFixup compiled from source
Hi how can I do this ?

baraach, 2021-10-31:
> Use the latest version of BlueToolFixup compiled from source

Latest compiled version of BlueToolFixup is causing bluetooth to stop working when open Bluetooth settings