OpenIntelWireless/IntelBluetoothFirmware #357: Bluetooth not working on macOS Monterey 12.0.1

KryptonGS, 2021-11-05:
**Description**

**Environment**

- Kext Version:  v2.0.1
- WiFi Card Model: AX200
- Product ID: 
- macOS Version: Monterey 12.0.1

IntelBluetoohInjector.kext doesn't work on macOS Monterey 12.0.1, takes too long to boot and shutdown, please rewrite Bluetooth stack.

sh0umik, 2021-11-05:
having the same problem. mine worked on big sur. After the upgrade it stopped working. @KryptonGS did you find any solution ?

KryptonGS, 2021-11-05:
Mine worked on Big Sur fine too, but I coudn't find a solution for Monterey. Apple changed the Bluetooh stack on Monterey so the BluetoothInjector.kext doesn't work anymore, it has to get rewritten so I guess there is no solution, until a new version of the kexts gets released :(

williambj1, 2021-11-05:
Read the docs

closed 2021-11-05 by williambj1