OpenIntelWireless/IntelBluetoothFirmware #358: Possible IntelBluetoothInjector.kext bug.  Extremely long boot times and missing internal drives after upgrade from Big Sur to Monterey.

dmcnaugh15, 2021-11-06:
**Description**

After upgrading to Monterey, had a couple problems, as follows:

1.     Extremely long boot times:  After selecting the appropriate drive from the boot picker, it took over 4 minutes to load to the mac screen (versus 16 seconds on Big Sur).  That's using the exact same EFI folder with OC 0.7.5 on the same computer.
2.     Internal hard drives were all missing (see screenshots).  Just as for number 1 (above), I was using the exact same EFI folder as for Big Sur, on the same computer.  Drives were visible for Big Sur, but not Monterey.

After disabling IntelBluetoothInjector.kext, the problem resolved.  Others have reported the same problem.  See this insanelymac thread: https://www.insanelymac.com/forum/topic/349510-missing-internal-drives-and-extremely-long-boot-times-after-monterey-upgrade-help/?tab=comments#comment-2771243

**Environment**
-Kext version: IntelBluetoothInjector.kext v2.0.1
- WiFi Card Model: ASUS ROG MAXIMUS XI HERO (WI-FI) - Z390 (Coffee Lake)
- Product ID: 
- macOS Version: 12.0 (Monterey)

**Logs**

http://DAMDAS.myqnapcloud.com:8082/share.cgi?ssid=cebeef9f960240328f7db0f5196af5c1

**Screenshots**
[image]
[image]

williambj1, 2021-11-06:
Read the  docs, this is a user mistake.

closed 2021-11-06 by williambj1

dmcnaugh15, 2021-11-06:
I see it now in the documentation. Thank you!