OpenIntelWireless/IntelBluetoothFirmware #360: When connecting to iPhone, pairing only works from Hack, then does not allow connection

zspherez, 2021-11-11:
**Description**
When connecting to my iPhone, I can only get them paired if I initiate from my hack, when I try to pair from iPhone I get a message that says "Pairing Unsuccessful, MacBook Pro is not supported." Image of this error is attached. When pairing from my hack, I can get them paired, but a connection can never be established between them.

**Environment**

- Kext Version: 2.0.1
- WiFi Card Model: Killer(R) Wi-Fi 6 AX1650s 160MHz Wireless Network Adapter (201D2W)
- Product ID: Vendor: Intel (8087), Product: (0026)
- macOS Version: 12.0.1

**Info in System Report - Bluetooth**

```
  Bluetooth Controller:
  Address:	AC:12:03:CE:CB:D7
  State:	On
  Chipset:	THIRD_PARTY_DONGLE
  Discoverable:	On
  Firmware Version:	v256 c256
  Product ID:	0x0001
  Supported services:	0x382039 < HFP AVRCP A2DP HID Braille AACP GATT Serial >
  Transport:	USB
  Vendor ID:	0x004C (Apple)
  Paired Bluetooth Devices:
  MX Master 3:
  Address:	DB:93:D8:83:05:3F
  Connected:	Yes
  Vendor ID:	0x046D
  Product ID:	0xB023
  Firmware Version:	MPM19.01_0015
  Minor Type:	Mouse
  Services:	0x400000 < BLE >
  iPhone12:
  Address:	90:81:58:38:0A:7B
  Vendor ID:	0x004C
  Product ID:	0x7412
  Firmware Version:	15.0.0
  Minor Type:	MobilePhone

Bluetooth HCI:

  Product ID:	0x0026
  Vendor ID:	0x8087  (Intel Corporation)
  Version:	0.02
  Speed:	Up to 12 Mb/s
  Location ID:	0x14500000 / 1
  Current Available (mA):	500
  Current Required (mA):	100
  Extra Operating Current (mA):	0
  Built-In:	Yes

```

**Logs**

[bluetooth.log](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/7521887/bluetooth.log)
```log
This command resulted in no logs, log grep bluetooth is attached as a file due to it being extensive.
```

**Kernel Panic Logs**

```
No KP
```

**Screenshots**

[image]

williambj1, 2021-11-11:
Why would you need this functionality? Nothing related is done in a firmware uploader, read the docs for more info.

closed 2021-11-11 by williambj1

zspherez, 2021-11-11:
test

zspherez, 2021-11-11:
@williambj1 the reasoning behind this is your comment on the itlwm issue [https://github.com/OpenIntelWireless/itlwm/issues/292#issuecomment-693728787](here) where you stated that one should confirm that connection between iPhone and hack can be established before attempting to troubleshoot AirDrop. I now see that you also stated AirDrop functionality is not expected, is this still the case?

williambj1, 2021-11-11:
Bluetooth is only one piece of the puzzle and apple does not pair two devices “explicitly” to establish a connection. No, Airdrop is still not expected, but it’s not caused by Bluetooth since people have reported that Airdrop works fine with a brcm wifi & intel Bluetooth combo.

zspherez, 2021-11-12:
Alright thank you, I'll take this issue over to the itlwm issues at some point then. Thanks!