OpenIntelWireless/IntelBluetoothFirmware #361: Bluetooth not working with error log: IOkit daemon (kernelmanagerd) stall 0 (60s) 'IOUSBHostDevice'

TomHeaven, 2021-11-17:
**Description**

Bluetooth not working with error log: 
I recently upgrade my Huawei Matebook 14 2020 I5 laptop from BigSur 11.6 to Monterey 12.0.1. I found that IntelBluetoothFirmware.kext causing slow boot with error log:
```
IOkit daemon (kernelmanagerd) stall 0 (60s) 'IOUSBHostDevice'
```
and the bluetooth device is no longer working in macOS Monterey 12.0.1 (empty bluetooth information in system information; bluetooth cannot be open in preferences tab). However, the bluetooth works fine in macOS BigSur.

I can confirm that the driver (com.zxystd.IntelBluetoothFirmware) is loaded using Kext Wizard. I tried to change kext versions but it does not help.

**Environment**

- Kext Version:  1.1.3 & 2.0.0 & 2.0.1
- WiFi Card Model: Intel Wireless AC 9560 【Works fine with AirportIntelItlwm.kext】
- Product ID: 0x0aaa
- macOS Version: 12.0.1

**Info in System Report - Bluetooth**

1.1.3 shows the following information: 
```
蓝牙控制器：
  地址：	NULL
  状态：	关闭
  芯片组：	BCM_4350C2
  可被发现：	关闭
  固件版本：	v0
  产品ID：	0x0001
  支持的服务：	0x382039 < HFP AVRCP A2DP HID Braille AACP GATT Serial >
  传输：	PCIe
  供应商ID：	0x004C (Apple)
```
2.0.0 & 2.0.1 just show the empty information.

**Logs**

The command  `log show --last boot | grep IntelFir` returns empty log.

**Kernel Panic Logs**
No panic occurs. It reports the following error when I load IntelBluetoothFirmware.kext with the injector kext.
```
 IOkit daemon (kernelmanagerd) stall 0 (60s) 'IOUSBHostDevice'
```

**Screenshots**

williambj1, 2021-11-17:
Read the docs.

closed 2021-11-17 by williambj1

TomHeaven, 2021-11-17:
Thanks. The docs fix the problem. I copy it here for those meet similar problems.

# What additional steps should I do to make Bluetooth work on macOS Monterey
+ Make sure IntelBluetoothFirmware.kext v1.1.3 or newer is used.
+ Exclude the injection of IntelBluetoothInjector.kext for macOS Monterey from the Bootloader's config
+ Include BlueToolFixup.kext from acidanthera/BrcmPatchRAM (opens new window)

PasTrik, 2022-02-21:
It's working for me (Monterey) thanks ❤️

trey-niu, 2023-04-06:
It solved the problem that macOS installer cannot find internal SSD and Bluetooth doesn't work. Many thanks.

hjw21century, 2023-05-02:
> Thanks. The docs fix the problem. I copy it here for those meet similar problems.
> 
> # What additional steps should I do to make Bluetooth work on macOS Monterey
> * Make sure IntelBluetoothFirmware.kext v1.1.3 or newer is used.
> * Exclude the injection of IntelBluetoothInjector.kext for macOS Monterey from the Bootloader's config
> * Include BlueToolFixup.kext from [acidanthera/BrcmPatchRAM](acidanthera/BrcmPatchRAM)

Very usefull.

klewisjohnson, 2023-08-14:
> 

so helpful thanks

GONEPOLY, 2024-09-08:
> Thanks. The docs fix the problem. I copy it here for those meet similar problems.
>
> # What additional steps should I do to make Bluetooth work on macOS Monterey
> * Make sure [...]

it has solved some problom but not all,thanks