OpenIntelWireless/IntelBluetoothFirmware #362: ax210，wfi works well，but Bluetooth is not wrok

zcgg2zhmm, 2021-11-23:
**Description**
ax210，wfi works well，but Bluetooth is not wrok

**Environment**

- Kext Version: 
- WiFi Card Model: ax210
- Product ID: 
- macOS Version: 12.1 beta3

**Info in System Report - Bluetooth**

```
Tue Nov 23 09:40:24.659 Usb Host Notification hostNotificationUSBDeviceInserted USB Receiver isApple N seqNum 1 Total 1
Tue Nov 23 09:40:24.661 Usb Host Notification hostNotificationUSBDeviceInserted Rapoo Gaming Keyboard isApple N seqNum 1 Total 2
Tue Nov 23 09:40:24.826 Usb Host Notification activated
Tue Nov 23 09:40:24.830 <airport[227]> _bsdDriver_init: registration for monitoring bssSteeringRequest event 
Tue Nov 23 09:40:24.830 
Tue Nov 23 09:40:24.830 <airport[227]> psCallback: set powersave(7) failed, 5 (Input/output error)
Tue Nov 23 09:40:24.830 <kernel> Not adding subscription due to missing boot-arg wlan.iorsubscriptions
Tue Nov 23 09:40:24.831 <kernel> Not adding subscription due to missing boot-arg wlan.iorsubscriptions
Tue Nov 23 09:40:24.831 <kernel>  Creating all peerManager reporters
Tue Nov 23 09:40:24.837 <airport[227]> _initLocaleManager: Started locale manager
Tue Nov 23 09:40:24.837 <airport[227]> _initInterface: Failed to query current SSID (Input/output error)
Tue Nov 23 09:40:24.837 <kernel>  IOCTL (from pid 227) not recognized: 383 out of 380
Tue Nov 23 09:40:24.837 Apple80211SetProperty: ioctl failed: bsdErr=-1, errno=102
Tue Nov 23 09:40:24.837 <kernel>  IOCTL (from pid 227) not recognized: 381 out of 380
Tue Nov 23 09:40:24.837 Usb Host Notification Error Apple80211Set: Input/output error seqNum 1 Total 2 chg 1 en1
Tue Nov 23 09:40:24.837 <airport[227]> _clamshellClosed: 'AppleClamshellState' not found
Tue Nov 23 09:40:24.852 <airport[227]> airportdProcessDLILEvent: <en1> event=9, flags=0xFFFF8863, lastPowerPref=on
Tue Nov 23 09:40:24.852 <airport[227]> airportdProcessDLILEvent: en1 attached (up)
Tue Nov 23 09:40:24.853 <airport[227]> _bsdDriver_init: registration for monitoring bssSteeringRequest event 
Tue Nov 23 09:40:24.853 
Tue Nov 23 09:40:24.853 <airport[227]> psCallback: set powersave(7) failed, 5 (Input/output error)
Tue Nov 23 09:40:24.853 <airport[227]> psCallback: set powersave(7) failed, 5 (Input/output error)
Tue Nov 23 09:40:24.854 <kernel> en1: setting diversity to: -1
Tue Nov 23 09:40:24.854 <kernel> en1: Error configuring antenna diversity (index = -1).
Tue Nov 23 09:40:24.854 <kernel> en1: setting tx antenna: -1
Tue Nov 23 09:40:24.854 <kernel> en1: Error configuring transmit antenna (index = -1).
Tue Nov 23 09:40:24.858 Usb Host Notification Error Apple80211Set: Operation not supported seqNum 2 Total 2 chg 1 p2p0
Tue Nov 23 09:40:24.858 <kernel>  IOCTL (from pid 227) not recognized: 363 out of 380
Tue Nov 23 09:40:24.859 <airport[227]> _bsdDriver_init: registration for monitoring bssSteeringRequest event 
Tue Nov 23 09:40:24.859 
Tue Nov 23 09:40:24.859 <airport[227]> psCallback: set powersave(7) failed, 5 (Input/output error)
Tue Nov 23 09:40:24.859 <airport[227]> psCallback: set powersave(7) failed, 5 (Input/output error)
Tue Nov 23 09:40:24.863 Usb Host Notification Error Apple80211Set: Operation not supported seqNum 3 Total 2 chg 1 awdl0
Tue Nov 23 09:40:24.863 <kernel>  IOCTL (from pid 227) not recognized: 363 out of 380
Tue Nov 23 09:40:24.863 Usb Host Notification Error Apple80211Set: Input/output error seqNum 4 Total 2 chg 1 en1
Tue Nov 23 09:40:24.893 <airport[227]> systemWokenByWiFi: System Wake Reason not found
Tue Nov 23 09:40:24.982 <airport[227]> ERROR: bluetoothd (161) is not entitled for com.apple.wifi.events, but allowing anyways for event type 1
Tue Nov 23 09:40:24.982 <airport[227]> ERROR: bluetoothd (161) is not entitled for com.apple.wifi.events, but allowing anyways for event type 5
Tue Nov 23 09:40:24.983 <airport[227]> ERROR: bluetoothd (161) is not entitled for com.apple.wifi.events.private, will not register for event type 130
Tue Nov 23 09:40:25.002 <airport[227]> ERROR: bluetoothd (161) is not entitled for com.apple.wifi.events.private, will not register for event type 100
Tue Nov 23 09:40:25.003 <airport[227]> ERROR: bluetoothd (161) is not entitled for com.apple.wifi.events.private, will not register for event type 122
Tue Nov 23 09:40:25.003 <airport[227]> ERROR: bluetoothd (161) is not entitled for com.apple.wifi.events.private, will not register for event type 123
Tue Nov 23 09:40:25.004 <airport[227]> ERROR: bluetoothd (161) is not entitled for com.apple.wifi.events, but allowing anyways for event type 5
Tue Nov 23 09:40:25.004 <airport[227]> ERROR: bluetoothd (161) is not entitled for com.apple.wifi.events, but allowing anyways for event type 7
Tue Nov 23 09:40:25.004 <airport[227]> ERROR: bluetoothd (161) is not entitled for com.apple.wifi.events, but allowing anyways for event type 1
Tue Nov 23 09:40:25.004 <airport[227]> ERROR: bluetoothd (161) is not entitled for com.apple.wifi.events, but allowing anyways for event type 2
Tue Nov 23 09:40:25.939 <airport[227]> systemWokenByWiFi: System Wake Reason not found
Tue Nov 23 09:40:26.060 <airport[227]> _processIPv4Changes: ARP/NDP offloads disabled, not programming the offload
Tue Nov 23 09:40:26.062 <airport[227]> _processIPv4Changes: ARP/NDP offloads disabled, not programming the offload
Tue Nov 23 09:40:31.695 <airport[227]> ERROR: SecurityAgent (372) is not entitled for com.apple.wifi.events.private, will not register for event type 100
Tue Nov 23 09:40:31.695 <airport[227]> ERROR: SecurityAgent (372) is not entitled for com.apple.wifi.events.private, will not register for event type 101
Tue Nov 23 09:40:31.695 <airport[227]> ERROR: SecurityAgent (372) is not entitled for com.apple.wifi.events, but allowing anyways for event type 7
Tue Nov 23 09:40:31.696 <airport[227]> ERROR: SecurityAgent (372) is not entitled for com.apple.wifi.events, but allowing anyways for event type 2
Tue Nov 23 09:40:31.696 <airport[227]> ERROR: SecurityAgent (372) is not entitled for com.apple.wifi.events, but allowing anyways for event type 1
Tue Nov 23 09:40:31.696 <airport[227]> ERROR: SecurityAgent (372) is not entitled for com.apple.wifi.events, but allowing anyways for event type 5
Tue Nov 23 09:40:31.696 <airport[227]> ERROR: SecurityAgent (372) is not entitled for com.apple.wifi.events, but allowing anyways for event type 6
Tue Nov 23 09:40:31.910 <airport[227]> systemWokenByWiFi: System Wake Reason not found
[... 521 lines cut ...]
Tue Nov 23 14:56:49.622 <airport[227]> ERROR: bluetoothd (2800) is not entitled for com.apple.wifi.events, but allowing anyways for event type 5
Tue Nov 23 14:56:49.622 <airport[227]> ERROR: bluetoothd (2800) is not entitled for com.apple.wifi.events.private, will not register for event type 130
Tue Nov 23 14:56:49.623 <airport[227]> ERROR: bluetoothd (2800) is not entitled for com.apple.wifi.events.private, will not register for event type 100
Tue Nov 23 14:56:49.624 <airport[227]> ERROR: bluetoothd (2800) is not entitled for com.apple.wifi.events.private, will not register for event type 122
Tue Nov 23 14:56:49.625 <airport[227]> ERROR: bluetoothd (2800) is not entitled for com.apple.wifi.events.private, will not register for event type 123
Tue Nov 23 14:56:49.625 <airport[227]> ERROR: bluetoothd (2800) is not entitled for com.apple.wifi.events, but allowing anyways for event type 5
Tue Nov 23 14:56:49.625 <airport[227]> ERROR: bluetoothd (2800) is not entitled for com.apple.wifi.events, but allowing anyways for event type 7
Tue Nov 23 14:56:49.625 <airport[227]> ERROR: bluetoothd (2800) is not entitled for com.apple.wifi.events, but allowing anyways for event type 1
Tue Nov 23 14:56:49.625 <airport[227]> ERROR: bluetoothd (2800) is not entitled for com.apple.wifi.events, but allowing anyways for event type 2

```

**Logs**

```log

```

**Kernel Panic Logs**

```

```

**Screenshots**

zxystd, 2021-11-23:
macOS 12 doesn't support BT5.2, ask apple for help.

closed 2021-11-23 by zxystd

zcgg2zhmm, 2021-11-23:
> macOS 12 doesn't support BT5.2, ask apple for help.

thanks a lot