OpenIntelWireless/IntelBluetoothFirmware #363: Montery IntelBluetoothFirmware not working

RNetc0rp, 2021-11-24:
**Description**

Wifi works no problem, System Report > USB does show BluetoothHCI.  However when attempting to access anything using bluetooth (Quick Setttings/Bluetooth or Settings > Bluetooth or AIrdrop) I only get the spinning beachball and then the program crashes. I have looked at the github docs page, and tried all steps mentioned their, including manually compiling the kext to no avail.  Additionaly I have tried editing the source to mask the VID and PID as the broadcom module that the bluetooth stack looks for.  Loading the kext from Library/Extensions or on OpenCore, Different versions also has no impact Thus far nothing has worked.  

Dmesg gives:
LibrarySandbox: bluetoothd deny mach-lookup com.apple.network.EAPOLControllerIOConsoleUsers

**Environment**

- Kext Version: v2.0.1
- WiFi Card Model: Intel AX201
- Product ID: 
- macOS Version: Mac OS Monterey Beta Build 21A5268h

**Info in System Report - Bluetooth**

```
  Bluetooth Controller:
  Address:	NULL
  State:	Off
  Discoverable:	Off
  Firmware Version:	v0 c0
  Product ID:	0x0001
  Supported services:	0x381939 < HFP AVRCP A2DP HID NtSv Game NtCl AACP GATT Serial >
  Transport:	UART
  Vendor ID:	0x004C (Apple)
```

**Logs**

```log
Attempted command, I got no result
```

**Kernel Panic Logs**

```
no kernel panics
```

**Screenshots**

Here is the relevant dmesg log section

[ 1952.720675]: **** [IOBluetoothFamily][AddHCIEventNotification] -- Added notification port for HIDShim
[ 1952.720711]: **** [IOBluetoothFamily][CreateHIDShimSharedMemoryRingBuffer] -- Created input shared memory for HIDShim
[ 1952.720738]: **** [IOBluetoothFamily][CreateHIDShimSharedMemoryRingBuffer] -- Created output shared memory for HIDShim
[ 1952.721229]: **** [IOBluetoothHCIUserClient][SetBTLPOffWL] -- mBluetoothFamily->mACPIMethods->SetBTLP (OFF) returned 0xE00002C7 error
[ 1952.721232]: **** [IOBluetoothHCIUserClient][SetBTLPOffWL] -- mBluetoothFamily->mACPIMethods->SetBTLP (OFF) returned 0xE00002C7 error
[ 1952.721263]: **** [IOBluetoothHCIUserClient][SetBTRSWL] -- mBluetoothFamily->mACPIMethods->SetBTRS() returned 0xE00002C7 error
[ 1952.721264]: **** [IOBluetoothHCIUserClient][SetBTRSWL] -- mBluetoothFamily->mACPIMethods->SetBTRS() returned 0xE00002C7 error
[ 1952.741502]: PMRD: setAggressiveness(0) kPMMinutesToSleep = 0
[ 1952.787732]: 65 duplicate reports for Sandbox: mobileassetd(291) deny(1) file-read-metadata /LibrarySandbox: **bluetoothd(1615) deny(1) mach-lookup com.apple.network.EAPOLControllerIOConsoleUsers: time(0) 0->0, lin 1, llk 0, **
[ 1953.724664]: IOConsoleUsers: gIOScreenLockState 1, hs 0, bs 0, now 0, sm 0x0
[ 1953.748946]: PMRD: setAggressiveness(0) kPMMinutesToSleep = 0
[ 1954.116395]: PMRD: power event 14 args 0 0x0
[ 1954.116417]: PMRD: wrangler RootDomainPower (ps 4, 0->1)
[ 1954.116421]: PMRD: evaluatePolicy( 0, 0x0 )
[ 1954.754960]: **** [IOBluetoothFamily][RemoveHCIEventNotification] -- Removed notification port for HIDShim
[ 1954.754969]: **** [IOBluetoothFamily][RemoveHIDShimSharedMemoryRingBuffers] -- Removed input shared memory for HIDShim
[ 1954.754973]: **** [IOBluetoothFamily][RemoveHIDShimSharedMemoryRingBuffers] -- Removed output shared memory for HIDShim
[ 1954.755151]: **** [IOBluetoothFamily][DetachUserClients] -- HIDShim userclient has been terminated

buraktamturk, 2021-11-24:
Read the docs again, it has a specific section about Monterey.

gabrie1lira, 2021-11-24:
I have the same problem after i upgraded to monterey. He neither finds nor is found.

williambj1, 2021-11-24:
Read the docs.

closed 2021-11-24 by williambj1

RNetc0rp, 2021-11-24:
With all do respect I have read the docs, If you are referring to the section about not using the IntelBluetoothInjector.kext and the Broadcom Bluetooth patch, I have already done that.
I have tried it with and without, The same error occurs regardless

RNetc0rp, 2021-11-24:
Is their another docs page besides https://openintelwireless.github.io/IntelBluetoothFirmware that I am missing?

williambj1, 2021-11-25:
> With all do respect I have read the docs, If you are referring to the section about not using the IntelBluetoothInjector.kext and the Broadcom Bluetooth patch, I have already done that.
> I have tried it with and without, The same error occurs regardless

Does `System Info - Bluetooth` still shows the transport as USB? If so, BluetoolFixup might not be loaded properly (check `kextstat` or bootloader logs to confirm).

Just to clarify, the issue was closed because it's out of scope for a firmware uploader.

RNetc0rp, 2021-11-25:
System Info - Bluetooth does show up as UART, 

also Bluetoolfixup is loaded.

[sh-3.2# kextstat | grep -i "bluetoolfixup"
Executing: /usr/bin/kmutil showloaded
No variant specified, falling back to release
   60    0 0                  0xa000     0xa000     as.acidanthera.BlueToolFixup (2.6.1) 6BD14135-1D87-386E-9A15-5321B9BFF102 <50 9 7 6 3 2 1>

I can understand if my situation is out of bounds of your github, can you advise me as to where I should pursue this issue?  I am somewhat new to the hackintosh scene.  I posted my problem on the IntelBluetoothFirmware page as that seems to be what errors out to me.

uttusharma, 2022-01-29:
> System Info - Bluetooth does show up as UART,
> 
> also Bluetoolfixup is loaded as far as I can tell.
> 
> [sh-3.2# kextstat | grep -i "bluetoolfixup" Executing: /usr/bin/kmutil showloaded No variant specified, falling back to release 60 0 0 0xa000 0xa000 as.acidanthera.BlueToolFixup (2.6.1) 6BD14135-1D87-386E-9A15-5321B9BFF102 <50 9 7 6 3 2 1>
> 
> I can understand if my situation is out of bounds of your github, can you advise me as to where I should pursue this issue? I am somewhat new to the hackintosh scene. I posted my problem on the IntelBluetoothFirmware page as that seems to be what errors out to me.

I had a similar issue. My issue was resolved by correct USB map. I was using USBMap.kext earlier but switching to [USBToolbox.kext  ](https://github.com/USBToolBox/kext). 
If issue presists try changing SMBIOS model.