OpenIntelWireless/IntelBluetoothFirmware #366: Не работает в bluetooch в mac os Monterey

Krsubba13, 2021-12-12:
- Kext Version: 2.0.1
- WiFi Card Model: ax210
- Product ID: asus vivobook x510unr
- macOS Version: 12.0.1

Не работает в bluetooch в mac os Monterey, также при добвлении kext bluetooch отваливаются usb ports

realgudv1n, 2021-12-12:
Привет! Bluetooth работает на macOS Monterey. Для этого необходимо убрать IntelBluetoothInjector.kext и добавить BlueToolFixup.kext (взять отсюда https://github.com/acidanthera/BrcmPatchRAM/releases/)

Krsubba13, 2021-12-12:
> Привет! Bluetooth работает на macOS Monterey. Для этого необходимо убрать IntelBluetoothInjector.kext и добавить BlueToolFixup.kext (взять отсюда https://github.com/acidanthera/BrcmPatchRAM/releases/ )

Это частично решило проблему,блютуз появился в системе и порты перестали отваливаться,но не видит никакие устройства и нельзя его теперь выключить:(
Толку от блютуза в монтерей теперь никакого:(

closed 2021-12-12 by williambj1

TheNexter, 2021-12-13:
> Это частично решило проблему,блютуз появился в системе и порты перестали отваливаться,но не видит никакие устройства и нельзя его теперь выключить:( Толку от блютуза в монтерей теперь никакого:(

Same :(